TheThingsArchive/ttn #534: Status feedback for gateway to router connections

uchobby, 2017-03-21:
This is a feature request/question for the Backend, Console, NOC and or integration.

- Explain what you want to do
Implement some status for gateway to router connections.

- Explain what steps you took
Can not locate any established method for confirming a working connection to a functional router.

- Explain what went wrong or what is missing
TTN routers seem to go down often without notice. There are many options besides the intended router for a gateway to connect with.  What we need is some feedback, like a ping, that can be used by the gateway to report when connections are not made and or other feedback to the operator that network services are down or impaired.

## Environment
Gateway software development.
I am using the developing code to run on a gateway device for LoRaWAN TTN.

massar, 2017-03-21:
A https://status.thethingsnetwork.org would be great to have....

uchobby, 2017-03-21:
Yes, something that list out each of the network systems and a real status.  This is essential for those of us developing on TTN.

What I am proposing is a bit different however. I want for the gateway to be able to determine if the messages it's receiving are in fact being delivered into a functional system.  Some way to confirm that messages have made it all the way to the final network server.

Might also be good if there was a path from the application layer back to the gateway to confirm that messages have actually made it to their final destination.  I could implement an MQTT message to the network server, that is forwarded to the gateway.

This can be implemented in the gateway with custom MQTT but that would bypass TTN all together.  We need an "official" solution to the status issue.

johanstokking, 2017-03-26:
Regarding your first point: to test connections between gateway and router, you can see the packet forwarder's log and see if you get the PUSH_ACK and PULL_ACK. We're working on a brand new packet forwarder written from scratch in Go, that has a real TCP connection with the router, so that makes life much easier.

Regarding your second and third point: tracing per gateway is already deployed. Register your gateway in the Console, and go to the Traffic view:

[image]

Also, you can see that your downlink has actually been sent. Next to that, there's LoRaWAN confirmed downlink support to also see whether your device has actually received it: 

[image]

uchobby, 2017-03-26:
What I'm proposing is a way for code in the gateway to determine if the system is healthy and functioning.  That it's actually comunicating with the system. 

I've had at least two debug sessions that were solved by changing the router IP Address due to some issue. And at least one where the network server was actually down.  Would have saved me a great many hours of work if my gateway could have informed me that it was not able to deliver into a healthy system.

johanstokking, 2017-03-26:
So you'll see no data coming in the Console, and if you do, you'll see why it does not arrive to your application if there are any addressing of key problems in between.

With the current packet forwarder you can see what's going on with logs, and with the new packet forwarder we will make diagnostics easier.

closed 2017-03-26 by johanstokking