TheThingsArchive/ttn #538: Support multicast

johanstokking, 2017-03-26:
This is a **feature request** for **the Backend**.

With multicast, we can send an unconfirmed downlink message with fixed gateway ID, `DevAddr`, `FCnt`, `NwkSKey` and `AppSKey`, without MAC commands, `ACK` and `ADRACKReq` cleared and with support for a queue (and thus `FPending`).

johanstokking, 2017-03-26:
A possible implementation is the Handler exposing a function to create a session, internally securely generating the two session keys and returning a session ID. The application can use this session ID to publish messages over MQTT or AMQP, with the session ID being part of the topic structure. The application can also delete a session. We might be able to reuse existing device functionality for sessions in the Broker and Network Server.

htdvisser, 2017-06-28:
- Blocked, because this requires Class C or Class B (issue #440 and #502)
- The current LoRaWAN specification indicates that the application layer is responsible for setting up multicast groups. This means that to the network, a multicast device will just be an ABP device that only receives downlink. 
- @johanstokking: Do we consider setting up a multicast group the responsibility of the Handler or of the application?

johanstokking, 2017-07-05:
Let's see what happens with standardizing multicast as part of future LoRaWAN spec.

When it gets standardized, I can imagine it becomes part of the Handler. For now, we'll leave it in an application that may use a package that handles all this.

closed 2018-10-05 by htdvisser