TheThingsArchive/ttn #539: Allow Application creation through Application Manager

joscha, 2017-03-27:
This is a **feature request** for **the Backend**.

- I'd like to be able to programmatically create applications
- Right now it seems the ApplicationManager only allows me to register existing applications with a handler.

## Environment

n/a

johanstokking, 2017-03-28:
@joscha the applications are managed by the Account Server, not by the Handler. Indeed, you need to register an existing application on the Handler.

For the Account Server, we also have an API:

https://www.thethingsnetwork.org/docs/network/account/api.html

Of course, you can also implement your own Account Server as long as it's compatible with this API.

closed 2017-03-28 by johanstokking

joscha, 2017-03-28:
Ah, didn't see that API, thank you. I am trying to create a terraform provider for TTN, so I needed the applications as well.

joscha, 2017-03-28:
> For the Account Server, we also have an API:
> 
> https://www.thethingsnetwork.org/docs/network/account/api.html

Are there go bindings available for this? I couldn't find any in https://github.com/TheThingsNetwork/ttn

htdvisser, 2017-03-28:
See https://github.com/TheThingsNetwork/go-account-lib