TheThingsArchive/ttn #541: no buildable Go source files

Yaoshicn, 2017-03-30:
This is a bug report for the Back-end. 

When I tried to install the gateway, there is a failure in "make dev-deps".
```
fetch "gopkg.in/sourcemap.v1/base64vlq"
fetch "gopkg.in/yaml.v2"
package google.golang.org/genproto/googleapis/api/annotations: unrecognized import path "google.golang.org/genproto/googleapis/api/annotations" (fetch https://google.golang.org/genproto?go-get=1: no go-import meta tag)
make: *** [dev-deps] Error 1
```

I tried "make dev-deps" again, and the output goes like this
```
govendor sync -v
utils/protoc-gen-ttndoc/types.go:11:2: no buildable Go source files in /Users/Yaoshi/Documents/go_workspace/src/github.com/grpc-ecosystem/grpc-gateway/third_party/googleapis/google/api
```

Would you please give some help on this issue?

Thanks in advance.

edwindvinas, 2017-03-30:
Hi Xuan,

I also experienced it before so what I did is I put a dummy .go program
such as doc.go with the package name on it. Now there is a buildable go
file and compile worked fine.

Example:
vi doc.go
------------
package api
//no codes needed
-----------

Save it and try compile again.

HTH,
Edwin

Warm Regards,

Edwin D. Vinas
https://edwin-daen-vinas.appspot.com
http://www.ulapph.com

htdvisser, 2017-03-31:
Thanks for reporting. Fixed it in 43711179a0c577823669857927f072f057453dd7

closed 2017-03-31 by htdvisser

Yaoshicn, 2017-03-31:
@edwindvinas Thank you for your suggestion.
@htdvisser Thank you for fixing the bug!

commit 43711179 (2017-03-31): Fix protoc-gen-ttndoc deps

Resolves #541