TheThingsArchive/ttn #543: Define correct precedence rules for gateway location

gonzalocasas, 2017-04-02:
This is a **feature request** for **the console**.

There are three potential sources of geo coordinates to place a gateway on the map:
 - Static local configuration: `local_config.json` using `"fake_gps": true`.
 - Dynamic local configuration using real GPS on the gateway.
 - Static remote configuration configured on the **Console**

Right now, it is confusing which one takes precedence if multiple of them are set. There is a description of the current flow in the forum by @jpmeijers: https://www.thethingsnetwork.org/forum/t/gateway-gps-configuration-in-local-conf-json/6176

Being able to complete the configuration of a gateway without the need to open the console is very nice, it makes it very self-contained, so it would be nice to keep. If the precedence rules of the coordinates from different sources are clear, it can be achieved, i.e.: values set in the console take precedence over local configuration.

And for greater flexibility, @jpmeijers suggested adding a checkbox in the console to say `Always use gateway reported coordinates` would allow it to work easily with dynamic local config (real gps), and it would also make the precedence rules very explicit in one sentence.

johanstokking, 2017-04-03:
Currently, the `fake_gps` has priority. But it's a nasty hack really.

The gateway connector protocol gets the location from the Account Server, set by the user in the Console, and uses that if there's no GPS.

When using the packet forwarder, the location set in the Console only updates the map, not the coordinates reported by the gateway.

htdvisser, 2017-05-26:
Here are the rules for gateway location:

For the gateway:

- If the gateway **has a GPS** and **has a GPS fix**:
  - The gateway **must** add the location calculated from GPS to status messages as `status.gps`
  - The gateway **must** add the `time` field to uplink messages as `message.gateway_metadata.gps.time`
  - The gateway **may** add the `gps` field to uplink messages as `message.gateway_metadata.gps`
- In all other cases:
  - The gateway **must not** add a location to the status messages.
  - The gateway **must not** add a `time` field to uplink messages.

For the backend:

- Each time the gateway sends a status message, a location is cached as "latest"
  - If the status message contains a location, the location from the status message **must** be cached.
  - If the status message does not contain a location, but the gateway location is set in its registration, the registered location **must** be cached.
  - In all other cases, the location **must** be stored as "empty".
- Each time the gateway sends an uplink message:
  - If the metadata of the uplink message contains a `gps` field, this field **must not** be changed
  - If the metadata of the uplink message does not contain a `gps` field **and** the "latest" status message from the gateway contains a location, the "latest" location **must** be inserted.
  - In all other cases, the location **must** remain "empty"

Note: The `fake_gps` option must only be used if the gateway has some other means to determine an accurate location

closed 2017-05-26 by htdvisser