TheThingsArchive/ttn #544: Move device to another application

jpmeijers, 2017-04-04:
This is a feature request for the Console.

When viewing a device's settings at
`https://console.thethingsnetwork.org/applications/<application_id>/devices/<device_id>/settings`
it would be useful to have a `Move to application` button. When clicked, a list of all the current user's applications should be listed. When a new application is chosen, the device's registration (and perhaps session too, if it is possible) should be moved to the new application.

allesklarbeidir, 2019-10-02:
Looks like there's no button yet, but is it possible to do so by any means?
If I register the device with all parameters being the same in a new application, nothing shows up.
The device address is generated automatically, so I can't set it to the same value.

johanstokking, 2020-02-11:
This is supported in V3 via device claiming. We won't support this in this repo, V2.

closed 2020-02-11 by johanstokking