TheThingsArchive/ttn #548: Broker and monitor build fail

rabiihlioui, 2017-04-07:
FAIL	github.com/TheThingsNetwork/ttn/api/broker
FAIL	github.com/TheThingsNetwork/ttn/api/monitor

johanstokking, 2017-04-07:
I can't reproduce. Can you give context (e.g. go version, branch, verbose build output, etc)

rabiihlioui, 2017-04-07:
after making the dependencies , i make the test by taping make test but it shows me that it failed to make the broker and the monitor , and i didn't know why ,i tried everything , the same error every time.
My go version is : go version go1.7.5 linux/386

This is what i got as output : 
dministrator@debian:~/work/src/github.com/TheThingsNetwork/ttn$ make test
go test ./amqp ./api ./api/broker ./api/discovery ./api/gateway ./api/handler ./api/monitor ./api/pool ./api/protocol/lorawan ./api/router ./cmd ./core/band ./core/broker ./core/component ./core/discovery ./core/discovery/announcement ./core/handler ./core/handler/application ./core/handler/device ./core/handler/functions ./core/networkserver ./core/networkserver/device ./core/proxy ./core/router ./core/router/gateway ./core/storage ./core/types ./mqtt ./ttnctl/util ./utils/fcnt ./utils/otaa ./utils/parse ./utils/security ./utils/toa
ok  	github.com/TheThingsNetwork/ttn/amqp	0.967s
ok  	github.com/TheThingsNetwork/ttn/api	0.454s
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x810e38c]

goroutine 25 [running]:
panic(0x8501720, 0x18b0c010)
	/usr/local/go/src/runtime/panic.go:500 +0x331
sync/atomic.AddUint64(0x18e5bf74, 0x1, 0x0, 0x1, 0x0)
	/usr/local/go/src/sync/atomic/asm_386.s:112 +0xc
github.com/TheThingsNetwork/ttn/api/broker.NewReferenceBrokerServer.func1(0x18e5bf40)
	/home/administrator/work/src/github.com/TheThingsNetwork/ttn/api/broker/reference_server.go:38 +0xaa
created by github.com/TheThingsNetwork/ttn/api/broker.NewReferenceBrokerServer
	/home/administrator/work/src/github.com/TheThingsNetwork/ttn/api/broker/reference_server.go:43 +0x1ae
FAIL	github.com/TheThingsNetwork/ttn/api/broker	0.344s
ok  	github.com/TheThingsNetwork/ttn/api/discovery	0.212s
ok  	github.com/TheThingsNetwork/ttn/api/gateway	0.186s
ok  	github.com/TheThingsNetwork/ttn/api/handler	0.094s
2017/04/07 09:14:39 transport: http2Server.HandleStreams received bogus greeting from client: "\x16\x03\x01\x00\x8c\x01\x00\x00\x88\x03\x03\a)͙\x0eo\x04\xe4Bby\xab8"
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x80f36cc]

goroutine 13 [running]:
panic(0x8519440, 0x18b0c010)
	/usr/local/go/src/runtime/panic.go:500 +0x331
sync/atomic.AddUint64(0x18e3b104, 0x1, 0x0, 0x1, 0x0)
	/usr/local/go/src/sync/atomic/asm_386.s:112 +0xc
github.com/TheThingsNetwork/ttn/api/monitor.NewReferenceMonitorServer.func1(0x18e3b0e0)
	/home/administrator/work/src/github.com/TheThingsNetwork/ttn/api/monitor/reference_server.go:42 +0x103
created by github.com/TheThingsNetwork/ttn/api/monitor.NewReferenceMonitorServer
	/home/administrator/work/src/github.com/TheThingsNetwork/ttn/api/monitor/reference_server.go:57 +0x24b
FAIL	github.com/TheThingsNetwork/ttn/api/monitor	0.572s
ok  	github.com/TheThingsNetwork/ttn/api/pool	1.494s
ok  	github.com/TheThingsNetwork/ttn/api/protocol/lorawan	0.112s
ok  	github.com/TheThingsNetwork/ttn/api/router	0.195s
ok  	github.com/TheThingsNetwork/ttn/cmd	0.856s
ok  	github.com/TheThingsNetwork/ttn/core/band	0.081s
ok  	github.com/TheThingsNetwork/ttn/core/broker	0.370s
ok  	github.com/TheThingsNetwork/ttn/core/component	3.861s

htdvisser, 2017-04-10:
If I look at the code this seems to be caused by

```go
atomic.AddUint64
```

From the `sync/atomic` docs (https://golang.org/pkg/sync/atomic/#pkg-note-BUG):

```
On x86-32, the 64-bit functions use instructions unavailable before the Pentium MMX.

On non-Linux ARM, the 64-bit functions use instructions unavailable before the ARMv6k core.

On both ARM and x86-32, it is the caller's responsibility to arrange for 64-bit alignment of 64-bit words accessed atomically. The first word in a global variable or in an allocated struct or slice can be relied upon to be 64-bit aligned.
```

As you seem to be using a 32 bit machine, I suspect the last point causes this issue, but don't understand why, because afaik these structs are aligned just fine:

broker:

```go
type metrics struct {
	uplinkIn   uint64
	downlinkIn uint64
}
```

monitor:

```go
type metrics struct {
	gatewayStatuses         uint64
	uplinkMessages          uint64
	downlinkMessages        uint64
	brokerUplinkMessages    uint64
	brokerDownlinkMessages  uint64
	handlerUplinkMessages   uint64
	handlerDownlinkMessages uint64
}
```

htdvisser, 2017-04-10:
The problem was that the `metrics` struct was embedded into another struct, leading to misaligned memory. Fixed in 2f4a06e3685486afb5d56352c19069fc87021223

closed 2017-04-10 by htdvisser

htdvisser, 2017-04-10:
PR #550 fixes a failing test on 386

commit 2f4a06e3 (2017-04-10): Ensure correct memory alignment for 32 bit systems

Resolves #548