TheThingsArchive/ttn #551: FAIL	github.com/TheThingsNetwork/core/discovery/announcement

rabiihlioui, 2017-04-11:
**# --- FAIL: TestDiscoveryAnnounce (0.02s)**
	doc.go:64: 
		/home/administrator/work/src/github.com/TheThingsNetwork/ttn/core/discovery/discovery_test.go:44
		Expected: nil
		Actual:   'dial tcp [::1]:6379: getsockopt: connection refused'
	doc.go:64: 
		/home/administrator/work/src/github.com/TheThingsNetwork/ttn/core/discovery/discovery_test.go:47
		Expected: nil
		Actual:   'dial tcp [::1]:6379: getsockopt: connection refused'
	doc.go:64: 
		/home/administrator/work/src/github.com/TheThingsNetwork/ttn/core/discovery/discovery_test.go:48
		Expected [] (length: 0) to have length equal to '1', but it wasn't!
panic: runtime error: index out of range [recovered]
	panic: runtime error: index out of range

goroutine 6 [running]:
panic(0x8691e20, 0x18d0c030)
	/usr/local/go/src/runtime/panic.go:500 +0x331
testing.tRunner.func1(0x18d0ba00)
	/usr/local/go/src/testing/testing.go:579 +0x14f
panic(0x8691e20, 0x18d0c030)
	/usr/local/go/src/runtime/panic.go:458 +0x40b
github.com/TheThingsNetwork/ttn/core/discovery.TestDiscoveryAnnounce(0x18d0ba00)
	/home/administrator/work/src/github.com/TheThingsNetwork/ttn/core/discovery/discovery_test.go:49 +0xb3d
testing.tRunner(0x18d0ba00, 0x87cc41c)
	/usr/local/go/src/testing/testing.go:610 +0x8c
created by testing.(*T).Run
	/usr/local/go/src/testing/testing.go:646 +0x2a5
**# FAIL	github.com/TheThingsNetwork/ttn/core/discovery	0.104s**
--- FAIL: TestCachedAnnouncementStore (0.02s)
	doc.go:64: 
		/home/administrator/work/src/github.com/TheThingsNetwork/ttn/core/discovery/announcement/cache_test.go:31
		Expected: nil
		Actual:   'dial tcp [::1]:6379: getsockopt: connection refused'
	doc.go:64: 
		/home/administrator/work/src/github.com/TheThingsNetwork/ttn/core/discovery/announcement/cache_test.go:39
		Expected: nil
		Actual:   'discovery-test-announcement-store:announcement:router:router1 not found'
	doc.go:64: 
		/home/administrator/work/src/github.com/TheThingsNetwork/ttn/core/discovery/announcement/cache_test.go:40
		Expected '<nil>' to NOT be nil (but it was)!
	doc.go:64: 
		/home/administrator/work/src/github.com/TheThingsNetwork/ttn/core/discovery/announcement/cache_test.go:47
		Expected: nil
		Actual:   'dial tcp [::1]:6379: getsockopt: connection refused'
	doc.go:64: 
		/home/administrator/work/src/github.com/TheThingsNetwork/ttn/core/discovery/announcement/cache_test.go:57
		Expected: nil
		Actual:   'dial tcp [::1]:6379: getsockopt: connection refused'
	doc.go:64: 
		/home/administrator/work/src/github.com/TheThingsNetwork/ttn/core/discovery/announcement/cache_test.go:69
		Expected: nil
		Actual:   'dial tcp [::1]:6379: getsockopt: connection refused'
	doc.go:64: 
		/home/administrator/work/src/github.com/TheThingsNetwork/ttn/core/discovery/announcement/cache_test.go:72
		Expected: nil
		Actual:   'discovery-test-announcement-store:app_eui:0102030405060708 not found'
	doc.go:64: 
		/home/administrator/work/src/github.com/TheThingsNetwork/ttn/core/discovery/announcement/cache_test.go:73
		Expected '<nil>' to NOT be nil (but it was)!
panic: runtime error: invalid memory address or nil pointer dereference [recovered]
	panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x80c9282]

goroutine 11 [running]:
panic(0x84e06e0, 0x18b0c010)
	/usr/local/go/src/runtime/panic.go:500 +0x331
testing.tRunner.func1(0x18b0b480)
	/usr/local/go/src/testing/testing.go:579 +0x14f
panic(0x84e06e0, 0x18b0c010)
	/usr/local/go/src/runtime/panic.go:458 +0x40b
github.com/TheThingsNetwork/ttn/core/discovery/announcement.TestCachedAnnouncementStore(0x18b0b480)
	/home/administrator/work/src/github.com/TheThingsNetwork/ttn/core/discovery/announcement/cache_test.go:74 +0x2252
testing.tRunner(0x18b0b480, 0x85de0ec)
	/usr/local/go/src/testing/testing.go:610 +0x8c
created by testing.(*T).Run
	/usr/local/go/src/testing/testing.go:646 +0x2a5
**# FAIL	github.com/TheThingsNetwork/ttn/core/discovery/announcement	0.154s**

htdvisser, 2017-04-11:
Looks like you skipped step 5 of the README.md:

> Make sure you have Redis and RabbitMQ installed and running.

closed 2017-04-11 by htdvisser

rabiihlioui, 2017-04-11:
thanks, it was my mistake.
I know this is too much but i need your help one more time. The mqtt fails, this is what the test shows me : 

ok  	github.com/TheThingsNetwork/ttn/core/discovery/announcement	0.114s
2017/04/11 06:53:47 proto: textWriter unindented too far
VALUE [72 105]
SIGQUIT: quit
PC=0x80a1307 m=0

goroutine 0 [idle]:
runtime.epollwait(0x4, 0xbfd0d314, 0x80, 0xffffffff, 0x0, 0xffffffff, 0x0, 0x0, 0x0, 0x2f5ef6, ...)
	/usr/local/go/src/runtime/sys_linux_386.s:538 +0x17
runtime.netpoll(0x8e3fa01, 0x0)
	/usr/local/go/src/runtime/netpoll_epoll.go:67 +0x8d
runtime.findrunnable(0x19120a00, 0x0)
	/usr/local/go/src/runtime/proc.go:2003 +0x55d
runtime.schedule()
	/usr/local/go/src/runtime/proc.go:2120 +0x202
runtime.park_m(0x1955eb40)
	/usr/local/go/src/runtime/proc.go:2183 +0x154
runtime.mcall(0xbfd0d9e8)
	/usr/local/go/src/runtime/asm_386.s:265 +0x47

goroutine 1 [chan receive, 9 minutes]:
testing.(*T).Run(0x1963f180, 0x893027b, 0xe, 0x8ac1228, 0x1910be01)
	/usr/local/go/src/testing/testing.go:647 +0x2c4
testing.RunTests.func1(0x1910be00)
	/usr/local/go/src/testing/testing.go:793 +0xa1
testing.tRunner(0x1910be00, 0x19147ee0)
	/usr/local/go/src/testing/testing.go:610 +0x8c
testing.RunTests(0x8ac2148, 0x8e3b3a0, 0x20, 0x20, 0x1962be01)
	/usr/local/go/src/testing/testing.go:799 +0x347
testing.(*M).Run(0x19147f84, 0x193731c0)
	/usr/local/go/src/testing/testing.go:743 +0x79
main.main()
	github.com/TheThingsNetwork/ttn/core/handler/_test/_testmain.go:116 +0x100

goroutine 17 [syscall, 9 minutes, locked to thread]:
runtime.goexit()
	/usr/local/go/src/runtime/asm_386.s:1612 +0x1

goroutine 11 [sleep]:
time.Sleep(0x540be400, 0x2)
	/usr/local/go/src/runtime/time.go:59 +0xd4
github.com/TheThingsNetwork/ttn/vendor/github.com/shirou/gopsutil/process.(*Process).Percent(0x1932db80, 0x540be400, 0x2, 0x0, 0x0, 0x0, 0x0)
	/home/administrator/work/src/github.com/TheThingsNetwork/ttn/vendor/github.com/shirou/gopsutil/process/process.go:141 +0x120
github.com/TheThingsNetwork/ttn/api/stats.init.1.func2()
	/home/administrator/work/src/github.com/TheThingsNetwork/ttn/api/stats/stats.go:41 +0x66
created by github.com/TheThingsNetwork/ttn/api/stats.init.1
	/home/administrator/work/src/github.com/TheThingsNetwork/ttn/api/stats/stats.go:47 +0x3f

goroutine 10 [sleep]:
time.Sleep(0x540be400, 0x2)
	/usr/local/go/src/runtime/time.go:59 +0xd4
github.com/TheThingsNetwork/ttn/vendor/github.com/shirou/gopsutil/cpu.Percent(0x540be400, 0x2, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0)
	/home/administrator/work/src/github.com/TheThingsNetwork/ttn/vendor/github.com/shirou/gopsutil/cpu/cpu.go:146 +0xdd
github.com/TheThingsNetwork/ttn/api/stats.init.1.func1()
	/home/administrator/work/src/github.com/TheThingsNetwork/ttn/api/stats/stats.go:28 +0x32
created by github.com/TheThingsNetwork/ttn/api/stats.init.1
	/home/administrator/work/src/github.com/TheThingsNetwork/ttn/api/stats/stats.go:34 +0x2a

goroutine 13 [chan receive]:
github.com/TheThingsNetwork/ttn/vendor/gopkg.in/redis.v5/internal/pool.(*ConnPool).reaper(0x195527e0, 0xf8475800, 0xd)
	/home/administrator/work/src/github.com/TheThingsNetwork/ttn/vendor/gopkg.in/redis.v5/internal/pool/pool.go:323 +0xaf
created by github.com/TheThingsNetwork/ttn/vendor/gopkg.in/redis.v5/internal/pool.NewConnPool
	/home/administrator/work/src/github.com/TheThingsNetwork/ttn/vendor/gopkg.in/redis.v5/internal/pool/pool.go:85 +0x1b4

goroutine 14 [chan receive]:
github.com/TheThingsNetwork/ttn/vendor/gopkg.in/redis.v5/internal/pool.(*ConnPool).reaper(0x19552930, 0xf8475800, 0xd)
	/home/administrator/work/src/github.com/TheThingsNetwork/ttn/vendor/gopkg.in/redis.v5/internal/pool/pool.go:323 +0xaf
[... 619 lines cut ...]
panic: Did not receive fields [recovered]
	panic: Did not receive fields

goroutine 594 [running]:
panic(0x8416f40, 0x18cd4808)
	/usr/local/go/src/runtime/panic.go:500 +0x331
testing.tRunner.func1(0x18c30600)
	/usr/local/go/src/testing/testing.go:579 +0x14f
panic(0x8416f40, 0x18cd4808)
	/usr/local/go/src/runtime/panic.go:458 +0x40b
github.com/TheThingsNetwork/ttn/mqtt.TestPublishUplinkFields(0x18c30600)
	/home/administrator/work/src/github.com/TheThingsNetwork/ttn/mqtt/uplink_test.go:104 +0xb46
testing.tRunner(0x18c30600, 0x8514c4c)
	/usr/local/go/src/testing/testing.go:610 +0x8c
created by testing.(*T).Run
	/usr/local/go/src/testing/testing.go:646 +0x2a5
# **FAIL	github.com/TheThingsNetwork/ttn/mqtt	86.245s**
ok  	github.com/TheThingsNetwork/ttn/ttnctl/util	0.116s
ok  	github.com/TheThingsNetwork/ttn/utils/fcnt	0.003s
ok  	github.com/TheThingsNetwork/ttn/utils/otaa	0.007s
ok  	github.com/TheThingsNetwork/ttn/utils/parse	0.003s
ok  	github.com/TheThingsNetwork/ttn/utils/security	0.045s
ok  	github.com/TheThingsNetwork/ttn/utils/toa	0.009s
Makefile:76: recipe for target 'test' failed
make: *** [test] Error 1

htdvisser, 2017-04-11:
Let me copy-paste the steps from README:

1. Make sure you have [Go](https://golang.org) installed (version 1.7 or later).
2. Set up your [Go environment](https://golang.org/doc/code.html#GOPATH)
3. Install the [protobuf compiler (`protoc`)](https://github.com/google/protobuf/releases)
4. Install `make`. On Linux install `build-essential`. On macOS, `make` comes with XCode or the developer tools. On Windows you can get `make` from [https://gnuarmeclipse.github.io/windows-build-tools/](https://gnuarmeclipse.github.io/windows-build-tools/)
5. Make sure you have [Redis](http://redis.io/download) and [RabbitMQ](https://www.rabbitmq.com/download.html) **installed** and **running**.  
  On a fresh installation you might need to install the [MQTT plugin for RabbitMQ](https://www.rabbitmq.com/mqtt.html).  
  If you're on Linux, you probably know how to do that. On a Mac, just run `brew bundle`. The Windows installer will setup and start RabbitMQ as a service. Use the `RabbitMQ Command Prompt (sbin dir)` to run commands, i.e. to enable plugins.
6. Declare a RabbitMQ exchange `ttn.handler` of type `topic`. Using [the management plugin](http://www.rabbitmq.com/management.html), declare the exchange in the web interface `http://server-name:15672` or using the management cli, run `rabbitmqadmin declare exchange name=ttn.handler type=topic auto_delete=false durable=true`. If your handler's user has sufficient permissions on RabbitMQ, it will attempt to create the exchange if not present.

rabiihlioui, 2017-04-11:
Thanks a lot , again my mistake , didn't enable the MQTT plugins.