TheThingsArchive/ttn #555: Extract Data from theThingsNetwork or Lora Gateway to my own Server

zaheentheawesome, 2017-04-12:
Can I have the method of extracting data from TheThingsNetwork and transfer them to my own server?
Is it possible that I don't use the TTN and send data to my server directly from LoraWan Gateway?

Kindly help

Zaheen
7042747307

johanstokking, 2017-04-12:
Yes, use [MQTT](https://www.thethingsnetwork.org/docs/applications/mqtt/) or the [HTTP integration](https://www.thethingsnetwork.org/docs/applications/http/).

You can also run the TTN open source components on your own server, see https://www.thethingsnetwork.org/article/setting-up-a-private-routing-environment

Finally, you can also implement your own LoRa(WAN) network server, starting here: https://github.com/Lora-net/packet_forwarder/blob/master/PROTOCOL.TXT

closed 2017-04-12 by johanstokking

zaheentheawesome, 2017-04-12:
Thanks for the information