TheThingsArchive/ttn #557: make dev-deps problem

rabiihlioui, 2017-04-19:
administrator@ubuntu:~/work/src/github.com/TheThingsNetwork/ttn$ make dev-deps
package bytes: unrecognized import path "bytes" (import path does not begin with hostname)
package flag: unrecognized import path "flag" (import path does not begin with hostname)
package fmt: unrecognized import path "fmt" (import path does not begin with hostname)
package bufio: unrecognized import path "bufio" (import path does not begin with hostname)
package errors: unrecognized import path "errors" (import path does not begin with hostname)
package io: unrecognized import path "io" (import path does not begin with hostname)
package os: unrecognized import path "os" (import path does not begin with hostname)
package strings: unrecognized import path "strings" (import path does not begin with hostname)
package syscall: unrecognized import path "syscall" (import path does not begin with hostname)
package unicode/utf8: unrecognized import path "unicode/utf8" (import path does not begin with hostname)
package unsafe: unrecognized import path "unsafe" (import path does not begin with hostname)
package strconv: unrecognized import path "strconv" (import path does not begin with hostname)
package encoding/json: unrecognized import path "encoding/json" (import path does not begin with hostname)
package crypto/sha1: unrecognized import path "crypto/sha1" (import path does not begin with hostname)
package encoding/base64: unrecognized import path "encoding/base64" (import path does not begin with hostname)
package crypto/rand: unrecognized import path "crypto/rand" (import path does not begin with hostname)
package encoding/base32: unrecognized import path "encoding/base32" (import path does not begin with hostname)
package path/filepath: unrecognized import path "path/filepath" (import path does not begin with hostname)
package runtime: unrecognized import path "runtime" (import path does not begin with hostname)
package log: unrecognized import path "log" (import path does not begin with hostname)
package time: unrecognized import path "time" (import path does not begin with hostname)
package sort: unrecognized import path "sort" (import path does not begin with hostname)
package path: unrecognized import path "path" (import path does not begin with hostname)
package encoding/xml: unrecognized import path "encoding/xml" (import path does not begin with hostname)
package os/exec: unrecognized import path "os/exec" (import path does not begin with hostname)
package sync: unrecognized import path "sync" (import path does not begin with hostname)
package io/ioutil: unrecognized import path "io/ioutil" (import path does not begin with hostname)
package go/ast: unrecognized import path "go/ast" (import path does not begin with hostname)
package go/parser: unrecognized import path "go/parser" (import path does not begin with hostname)
package go/printer: unrecognized import path "go/printer" (import path does not begin with hostname)
package go/token: unrecognized import path "go/token" (import path does not begin with hostname)
package net/http: unrecognized import path "net/http" (import path does not begin with hostname)
package net/url: unrecognized import path "net/url" (import path does not begin with hostname)
package regexp: unrecognized import path "regexp" (import path does not begin with hostname)
package hash: unrecognized import path "hash" (import path does not begin with hostname)
package math: unrecognized import path "math" (import path does not begin with hostname)
package unicode: unrecognized import path "unicode" (import path does not begin with hostname)
package text/tabwriter: unrecognized import path "text/tabwriter" (import path does not begin with hostname)
package text/template: unrecognized import path "text/template" (import path does not begin with hostname)
make: *** [build-deps] Error 1

go version 1.8

Sypheos, 2017-04-19:
Hi there !

Can you provide more info ? Look like it come from the go installation.

rabiihlioui, 2017-04-19:
problem solved by Add govendor to deps #521

Sypheos, 2017-04-19:
Did you installed govendor ?

go get -u github.com/kardianos/govendor

rabiihlioui, 2017-04-19:
i changed i line in makefile, it didn't know the whole path of the command govendor

Sypheos, 2017-04-19:
Don't forget to set $GOBIN and append it to your $PATH variable.

https://golang.org/doc/code.html#GOPATH

closed 2017-04-19 by Sypheos