TheThingsArchive/ttn #562: Could not connect to MQTT Broker

rabiihlioui, 2017-04-19:
After running forego start i get a failure saying that i can not connect to the mqtt brocker.

.
.
.
hdl.1  |   INFO ttn: Announced to TTN discovery         
hdl.1  |   WARN Could not connect to MQTT Broker (Network Error : dial tcp 127.0.0.1:1883: getsockopt: connection refused). Retrying...
rtr.1  |   INFO ttn: Got public keys for token validation
.
.
.
.
=dev Method=/discovery.Discovery/Get ServiceName=broker ServiceVersion=v2.6.1-dev-7da6ca59adfe5773a19f54ee2483b18e2ea38625 (2017-04-19T11:05:38Z)
brk.1  |   WARN Prefix not announced in Discovery        Prefix=26000000/20
hdl.1  |   WARN Could not connect to MQTT Broker (Network Error : dial tcp 127.0.0.1:1883: getsockopt: connection refused). Retrying...
hdl.1  |   WARN Could not connect to MQTT Broker (Network Error : dial tcp 127.0.0.1:1883: getsockopt: connection refused). Retrying...
hdl.1  |   WARN Could not connect to MQTT Broker (Network Error : dial tcp 127.0.0.1:1883: getsockopt: connection refused). Retrying...
hdl.1  |   WARN Could not connect to MQTT Broker (Network Error : dial tcp 127.0.0.1:1883: getsockopt: connection refused). Retrying...
hdl.1  |   WARN Could not connect to MQTT Broker (Network Error : dial tcp 127.0.0.1:1883: getsockopt: connection refused). Retrying...
hdl.1  |   WARN Could not connect to MQTT Broker (Network Error : dial tcp 127.0.0.1:1883: getsockopt: connection refused). Retrying...
hdl.1  |   WARN Could not connect to MQTT Broker (Network Error : dial tcp 127.0.0.1:1883: getsockopt: connection refused). Retrying...
hdl.1  |   WARN Could not connect to MQTT Broker (Network Error : dial tcp 127.0.0.1:1883: getsockopt: connection refused). Retrying...
hdl.1  |   WARN Could not connect to MQTT Broker (Network Error : dial tcp 127.0.0.1:1883: getsockopt: connection refused). Retrying...
hdl.1  |  FATAL Could not initialize handler             error=Could not connect to MQTT Broker (Network Error : dial tcp 127.0.0.1:1883: getsockopt: connection refused)
forego | sending SIGTERM to rtr.1
forego | sending SIGTERM to ns.1
forego | sending SIGTERM to brk.1
forego | sending SIGTERM to dsc.1
rtr.1  |   INFO signal received                          signal=terminated
ns.1   |   INFO signal received                          signal=terminated
brk.1  |   INFO signal received                          signal=terminated
dsc.1  |   INFO signal received                          signal=terminated

htdvisser, 2017-04-20:
The discovery component (dsc) runs on port 1900. While the discovery server is still starting, you might see these messages (note that they're just DEBUG messages). If this keeps happening, check why the discovery component is not starting.

htdvisser, 2017-04-20:
closing for now

htdvisser, 2017-04-20:
Also, please follow the issue template. It's there for a reason.

closed 2017-04-20 by htdvisser