TheThingsArchive/ttn #563: Changing device description of an OTAA device, system returns to pre-join state for device

forrestfiller, 2017-04-19:
This is a **bug report** for **the Console**

- When changing the device description of an OTAA device, the system returns to the pre-join state for that OTAA device. As a result:
  - The device's `DevAddr` and `App Session keys` are no longer valid, thus the device must be manually instructed to re-join.

This was discovered by @terrillmoore and I tonight at the [Heat Seek](http://heatseek.org/) meetup as we work to set them up to pilot with TTN.

Thanks!

Forrest

Please tag [Terry Moore](https://github.com/terrillmoore) in all correspondence.

htdvisser, 2017-04-20:
Confirmed

[image]

closed 2017-04-20 by romeovs

htdvisser, 2017-04-20:
Fixed in console repository. Will be deployed to production soon.

htdvisser, 2017-04-20:
Deployed to production. Fix confirmed.

[image]

terrillmoore, 2017-04-20:
Thanks for the prompt response, all! Much appreciated!

forrestfiller, 2017-04-22:
Yes, thanks again, appreciate it! :+1: