TheThingsArchive/ttn #564: Current TTN USERNAME does not allow for any valid email address

OwenBrotherwood, 2017-04-20:
It is often desirable to use one's email address as a USERNAME.
The current check on valid USERNAME does not allow this.
Example:
name.notethedot@domain.tl
ie the . (full stop/dot) is not allowed

[image]

johanstokking, 2017-04-20:
We need a username for public display (e.g. on the forum, community pages, gateway info). E-mail addresses are kept private, unless you decide otherwise.

The reason why you can't use e-mail address characters for the username, is that we want the username to be used in urls as well.

closed 2017-04-20 by johanstokking

OwenBrotherwood, 2017-04-20:
Probably a good place to state email not advised ...

```
USERNAME
This will be your username — pick a good one because you will not be able to change it.
```