TheThingsArchive/ttn #567: Use Let's Encrypt Certificates on MQTT Brokers

htdvisser, 2017-04-21:
Currently the public MQTT brokers use TLS certificates that are signed by a CA that is not globally trusted (The Things Network CA). This requires MQTT clients to download the root cert in order to be able to connect to MQTT.

If we'd use Let's Encrypt certificates, many clients would already trust the server because Let's Encrypt is now in the OS's list of trusted roots.

We can easily start doing this for the MQTT brokers on the gateway side, but for the application side, we have to come up with some way to transition to Let's Encrypt certificates without breaking existing clients.

Note that the existing (temporary, TTN CA signed) MQTT certs expire in August this year

htdvisser, 2017-05-12:
Let's Encrypt is now enabled on all gateway-facing MQTT brokers.

hoonppark, 2017-05-12:
Do I need to set up the Let's Encrypt certificates for MQTT brokers that are receiving packets via the TTN-gateway-bridge?  A MQTT broker without a server certificate is currently working fine with the TTN-gateway-bridge.   Is it required only for gateways using the new gateway protocol using MQTT?

hoonppark, 2017-05-12:
I have an additional question regarding the certificate's valid period.
Let's Encrypt certificates are valid for 90 days.  Do you plan to restart all MQTT brokers when you renew the Let's Encrypt certificates every 60 days or 80 days?

htdvisser, 2017-05-13:
Regarding your first question: You don't _have_ to set up Let's Encrypt certificates (or any certificates at all). Plain MQTT (without TLS) is available on port 1883. TLS is on port 8883. Almost all gateway implementations that use MQTT do not have TLS enabled yet, so nothing has to be changed there.

Regarding your second question: correct, we make sure the MQTT brokers automatically reload their config after the certificates have been renewed.

htdvisser, 2017-08-02:
Done

https://status.thethings.network/incident/14
https://www.thethingsnetwork.org/forum/t/secure-mqtt-apis-now-using-lets-encrypt-certificates/8774

closed 2017-08-02 by htdvisser