TheThingsArchive/ttn #568: Some issue in ADR algorithm implementation

mariuszslabicki, 2017-04-21:
This is a **probably bug** report for ADR algorithm. 

I want to evaluate ADR in simulating environment.

In the last loop (lines: 64-66 in adr.go file) there is mechanism which increase TxPower every nStep:
```
	for nStep < 0 && txPower < f.ADR.MaxTXPower {
		txPower += 3
	}
```

But this nStep variable is not incremented in the loop. So the final output is that the txPower will be set for f.ADR.MinTXPower, because loop won't stop before that.

Link to file:
https://github.com/TheThingsNetwork/ttn/blob/develop/core/band/adr.go
From commit: bbde582

htdvisser, 2017-04-21:
Thanks for your report. This indeed seems to be wrong. Do you want to submit a pull request with a fix? Otherwise I'll fix this next week.

mariuszslabicki, 2017-04-24:
Done. It was a simple change.

htdvisser, 2017-04-24:
I know 😉 but a simple change that just made you a contributor to this awesome project 👍

mariuszslabicki, 2017-04-24:
Thank you, I am glad if I helped.

closed 2017-04-25 by htdvisser