TheThingsArchive/ttn #569: Local endpoint for the things network account servers

rabiihlioui, 2017-04-21:
Can i use my own endpoint instead of using https://account.thethingsnetwork.org. I want to run the backend without connecting to you server ( a total local network ).

htdvisser, 2017-04-24:
Yes. You can implement your own account server. If it follows the API, it will work just fine. See [this documentation page](https://www.thethingsnetwork.org/docs/network/account/implement.html) for more information.

closed 2017-04-24 by htdvisser

mzacharioudakis, 2018-03-02:
Hello, I have a question on this. The flow is that the user authenticates himself through the private account server. After that he/she wants to see their registered applications. Which component (and api) the account server must call ?
More specific - these [apis](https://www.thethingsnetwork.org/docs/network/account/api.html#endpoints) from which component are offered ? Are they prebuild or we have to develop them?

htdvisser, 2018-03-02:
https://www.thethingsnetwork.org/docs/network/account/implement.html lists the required endpoints. However, I think you'd save yourself a lot of trouble if you'd just use the public account server or wait for our v3 backend.

mzacharioudakis, 2018-03-02:
@htdvisser thank you for your reply. I want to let the user create an application through a gui. I want to consume these apis ([see this link plz](https://www.thethingsnetwork.org/docs/network/account/api.html)). Could you tell me which component have them?

htdvisser, 2018-03-02:
Those endpoints are exposed by the Account Server under https://account.thethingsnetwork.org/api/v2

mzacharioudakis, 2018-03-02:
@htdvisser, I am talking about a private solution. 
I have the account server which authenticates the user. However, I want to let the user register (create) an application. In which component I have to POST the information about the new application? If these apis are exposed by the account server, then the information is stored in the account server's database. How the backend is informed about that?

johanstokking, 2018-03-04:
Indeed, this is in the Account Server's database.

You can register applications in the backend through `ttnctl applications register ...`.

Note that in TTN Stack V3, the Identity Server (successor to Account Server) will be open source.