TheThingsArchive/ttn #572: Send component status messages to the NOC

rvolosatovs, 2017-04-24:
This is a **feature request** for **the backend**.

Router, handler and broker status messages should be sent to the NOC.

htdvisser, 2017-04-24:
Let's also add the networkserver here

htdvisser, 2017-04-24:
This will be streams of

- [`router.Status`](https://github.com/TheThingsNetwork/ttn/blob/develop/api/router/router.proto#L89-L101)
- [`broker.Status`](https://github.com/TheThingsNetwork/ttn/blob/develop/api/broker/broker.proto#L168-L182)
- [`networkserver.Status`](https://github.com/TheThingsNetwork/ttn/blob/develop/api/networkserver/networkserver.proto#L49-L58)
- [`handler.Status`](https://github.com/TheThingsNetwork/ttn/blob/develop/api/handler/handler.proto#L37-L44)

htdvisser, 2017-06-06:
@gomezjdaniel: please discuss with @rvolosatovs if component statuses can already be sent to the NOC. If ok, please make sure the routing components send their status to the NOC every X seconds. 

For the Handler, it's probably just a `time.Tick` and `monitorStream.Send(h.GetStatus())`, but for the Router and Networkserver you might have to add some initialization (just copy-paste)

htdvisser, 2017-06-06:
@jankramer: Does vx support the component status RPCs?

jankramer, 2017-06-06:
@htdvisser no, the plan is to get these from the NOC instead.

rvolosatovs, 2017-06-06:
The component statuses can be sent to the NOC once @htdvisser approves TheThingsIndustries/noc#197 and I deploy
They will not get stored yet though (that's TheThingsIndustries/noc#190)

closed 2017-06-28 by htdvisser