TheThingsArchive/ttn #573: Add and Delete application error

rabiihlioui, 2017-04-24:
This is a **{bug report/feature request/question/...}** for **{the website/the backend/ttnctl/the console/the NOC/an integration}**.

- Explain what you want to do
- Explain what steps you took
- Explain what went wrong or what is missing

## Environment

- `ttn version` returns: `Commit={...}`
- `ttnctl version` returns: `Commit={...}`
- My application:
  - App ID: `{...}`
- My gateway:
  - EUI: `{...}`
  - Connected to: `router.{...}.thethings.network:1700`
- My node:
  - Device ID: `{...}`
  - AppEUI: `{...}`
  - DevEUI: `{...}`
- MQTT is connected to: `{...}.thethings.network:{1883/8883}`
- I am using the `{...}` library for the `{...}` programming language

Sypheos, 2017-04-24:
Application are actually not deleted. When you 'delete' they are hidden and not accessible anymore.
It's confusing and we are working on it.

For the moment I would suggest you stick with the App you have and only remove/add devices inside.

htdvisser, 2017-04-25:
@romeovs: 

- Please return OK when deleting an already deleted resource
- Please allow re-registering an application if the owner is the same

htdvisser, 2017-04-25:
@rabiihlioui:

Please follow the issue template when creating issues. That really helps us when we address them.

romeovs, 2017-05-05:
This is actually a very subtle issue and resolving it should be done with great care. 

Going at it I came up with a couple of not-so-trivial questions:

1. Should all collaborators be able to restore an app? If so what rights should they have? `delete` presumably. But then we can't remove the app collaborators from the app on deletion. What if someone restores the app without our consent? This might introduce a lot of unexpected behavior for user (who generally would expect deleting an app removes all it's state, mainly the collaborators).
2. Note that an app does not know a concept of *owner* it just has a number of collaborators. We could assume it's the first user that created the app, but that's fragile and introduces a lot of support requests (like the gateway already has):
>  please change the owner of app `foo` to my colleague
Unless we also create an api for this (maybe just a right `restore` might work here).
3. Deleting an app removes it from all future jwt access tokens users create (that's the whole point), but how to authenticate for restoring an app then? You can never obtain authentication for a deleted app. We would need to build a custom authentication step for this, further complicating the account server api.

cc @htdvisser @johanstokking

johanstokking, 2017-05-05:
Let's not make it too complicated by introducing an undelete feature. That is asking for trouble as you illustrated. Also constraining only old collaborators with the `delete` right to undelete the application is not intuitive.

I think there are two options:
1. There's no such thing as deleting

htdvisser, 2017-05-05:
Let's just call it "Archive" instead of delete and be done with it

ama9910, 2017-08-21:
The issue I observed is that when deleting an application with ttnctl, the devices remained active. There is now no way to delete those devices using ttnctl, without first selecting the now deleted application - which can't be done.

Proposed option 1 above would allow the application to be unarchived and the devices deleted.

Another alternative might be for `ttnctl applications delete` also delete the devices as @htdvisser suggested in #private-backend.

johanstokking, 2017-08-21:
Yes, these are two separate things. We will take unarchiving into account for the next version of the account server. When an application is deleted, it should be unregistered from the handler, which invokes deleting devices.

closed 2018-10-05 by htdvisser