TheThingsArchive/ttn #574: Could not set device on handler

rabiihlioui, 2017-04-24:
administrator@ubuntu:~$ ttnctl devices register test
  INFO Using Application                        AppEUI=70B3D57EF0004845 AppID=omniacom4
  INFO Generating random DevEUI...             
  INFO Generating random AppKey...             
  INFO Discovering Handler...                   Handler=dev
  INFO Connecting with Handler...               Handler=localhost:1904
 FATAL Could not register Device                error=Could not set device on Handler: Application not registered to this Handler: handler:application:omniacom4 not found source=Wrap: /home/administrator/work/src/github.com/TheThingsNetwork/ttn/utils/errors/errors.go:217
administrator@ubuntu:~$

Sypheos, 2017-04-24:
This error appear because register to register a device you must pass via the App it belong to and to modify anything in an App you need to register it in the handler.

**Select your application**
`ttnctl applications select <App_ID>`

**Register your application in the handler**
`ttnctl applications register`

rabiihlioui, 2017-04-24:
thanks a lot, another thing, i didn't know how to see the payload i'm sending to my server, i wonna see them on my terminal.

Sypheos, 2017-04-24:
What do you mean ?

htdvisser, 2017-04-25:
Hi @rabiihlioui,

Please follow the issue template when creating issues. That really helps us when we address them.

closed 2017-04-25 by htdvisser

rabiihlioui, 2017-04-26:
I'm sending payloads from a device to a my local server and i wonna see them on the terminal to make sure i'm recieving 'em, just like the payloads' table of the device in the consol.