TheThingsArchive/ttn #577: Handler not discovered in local server

rabiihlioui, 2017-04-26:
root@debian:/home/administrator/ttn# ./ttnctl applications register
  INFO Discovering Handler...                   Handler=mynetwork-handler
  INFO Connecting with Handler...               Handler=localhost:1904
 FATAL Could not register application           error=Could not register application on Handler: Application already exists source=Wrap: /go/src/github.com/TheThingsNetwork/ttn/utils/errors/errors.go:217

htdvisser, 2017-04-26:
@rabiihlioui: Please follow the issue template when creating issues.

closed 2017-04-26 by htdvisser

rabiihlioui, 2017-04-26:
didn't understand you

htdvisser, 2017-04-26:
When creating issues, we require everyone to follow the issue format, so that we can reproduce it:

[image]

rabiihlioui, 2017-04-26:
ok i'm sorry.