TheThingsArchive/ttn #593: TTN integration: more advanced config tempaltes

cambierr, 2017-05-05:
This is a **feature suggestion** for the **integration controller**.

What I'm suggesting is a more advance configuration template for the integration system.

We could build it so that we have html selects with predefined values (example of the HTTP integration having the HTTP methods).

Also, we would have conditional sub-fields. Again, with the HTTP integration sample: in case the POST method is chosen, we could offer a new select where the user would be able to choose between sending data via url vars, or via the actual POST body.

I understand this could take some time to implement but I think that on the long run, this could bring a lot of user-friendliness to the integration system.

johanstokking, 2017-05-05:
Good one @cambierr. It's indeed a lot of work and a sliding edge to support everything.

I'd say there's a difference between data types (which could be an enum) and conditional fields. The former is pretty easy, the latter is the challenge mostly.

Do you know of a good practice for doing this?

cambierr, 2017-05-05:
Not really, since 99% of such things are in web UI, written in tailor-made JS.

We could define a `default-visibility` property on all `ProcessSetting` objects.

Then, for the enum values, having and `action` objects containing `hide` and `display` properties being list of field names to be shown/hidden ? 

Keeping the HTTP example, this would give us something like:
```
settings: 
  - 
    defaultValue: GET
    defaultVisibility: display
    description: "Http method used"
    key: method
    name: Method
    required: true
    type: enum
    values: 
      - 
        key: GET
        name: GET
        onSelected: 
          - 
            hide: 
              - post-method
      - 
        key: POST
        name: POST
        onSelected: 
          - 
            display: 
              - post-method
  - 
    defaultValue: url
    defaultVisibility: hide
    description: "how the data will be sent"
    key: post-method
    name: "Post Method"
    required: true
    type: enum
    values: 
      - 
        key: url
        name: URL
      - 
        key: body
        name: Body

```
Not sure about how nice this is but it's the first thing that came to my mind!

johanstokking, 2017-05-05:
@romeovs what do you think?

romeovs, 2017-05-08:
I think we should try to avoid rolling our own validation format/library. Also I would try to keep the UI-related stuff out of the options of the integration controllers.

JSON-Schema could be a nice addition to how we define the options as it can express dependencies between options.

So for example the HTTP integration could have options like:
```
{
	"$schema": "http://json-schema.org/draft-04/schema#",
	"type": "object",
    "oneOf": [{
    	"type": "object",
      	"required": [ "method" ],
		"properties": {
			"method": {
          		"type": "string",
          		"enum": [ "GET" ],
            }
        }, 
    },{
    	"type": "object",
    	"required": [ "method", "post-method" ],
		"properties": {
			"method": {
          		"type": "string",
          		"enum": [ "PUT", "POST" ],
            },
          	"post-method": {
              "type": "string",
              "enum": [ "body", "url-query" ],
            },
        }
    }]
}
```

This encapsulated the validation perfectly.

I'm aware that it's not super-clear that the what the interdependencies are here, but it should be possible to build a generic UI on top of this.

closed 2021-06-03 by htdvisser