TheThingsArchive/ttn #594: Console: adding new gateway with gateway connector leaves router empty

kersing, 2017-05-07:
When adding a new gateway to the TTN console with the 'gateway connector' the router value is not on the list of required fields. It has to be set in the properties screen.
IMHO this should be a required value to enter on the add gateway screen like the frequency plan.

htdvisser, 2017-05-08:
Please follow the issue template when submitting issues. That saves us a lot of time.

This is an **improvement** for **the console**.

> What do you want to do?

Configure a new gateway running the multi-protocol-forwarder ("gateway connector" )

> What steps did you take?

- Go to the console
- Go to gateways
- "Register gateway"
- Fill all fields
- Submit

> What went wrong or what is missing?

- The "router" value can not be set

> Do you have Screenshots?

...

> What kind of OS/Browser/Gateway are you using? Which version?

...

> What are the IDs and EUIs of your Device/Gateway? (if applicable)

...

> What do your configuration files look like?

...

> What do your log files look like?

...

> Can you fix this yourself and submit a pull request?

No (console)

jpmeijers, 2017-05-08:
This is a bug report for the console.

Steps to reproduce:
1. Log into the console and register a new gateway. Choose the frequency pla. (ex. gateway id `test123`, using `EU_863_870`)
2. Got to https://account.thethingsnetwork.org/gateways/test123.
3. You can see the frequency plan is set, but the router field is empty.

Expected behaviour:
The console should automatically fill the router field according to the frequency plan chosen. For `EU_862_870` it should be router.eu.thethings.network.

romeovs, 2017-05-23:
This is fixed. The router field is now mandatory in the Console.

closed 2017-05-23 by romeovs