TheThingsArchive/ttn #604: Zero coordinates not replaced by console coordinates

jpmeijers, 2017-05-15:
This is a bug report for the backend.

> What do you want to do?

On some gateways you can not disable "fake gps", and the workaround is to set the coordinates to lat=0 and lon=0.

> What steps did you take?

Using the packet forwarder from: https://github.com/tftelkamp/single_chan_pkt_fwd
Set the coordinates to zero:
```
// Set location
float lat=0.0;
float lon=0.0;
int   alt=10;
```
Run the packet forwarder and the packets that are sent out looks like this:
```
{"stat":{"time":"2017-05-15 14:29:06 GMT","lati":0.00000,"long":0.00000,"alti":10,"rxnb":0,"rxok":0,"rxfw":0,"ackr":0.0,"dwnb":0,"txnb":0,"pfrm":"Single Channel Gateway","mail":"foo@bar.com","desc":"My gateway"}}
```

On the NOC you see:
```{"id":"eui-xxx","status":{"time":"xxx","gps":{"latitude":0.0000012673606,"longitude":0.000002682209,"altitude":10}```

> What went wrong or what is missing?

Because I sent "0.00000" rather than "0" as the lat and lon, it gets rounded off and not interpreted as "Null".
The coordinates are then not replaced by the values from the Console.

After we forced the gateway to send coordinates as "0" and not "0.00000" (what sprintf gives out), it worked fine.

> Can you fix this yourself and submit a pull request?

In this specific case it was fixed on the gateway side, but because it is an issue that can happen in other software too, it will be better to also fix this in the bridge/router.

> Recommendation

I think it would be a good idea to filter all invalid coordinates, or coordinates that are likely incorrect and replace them by the coordinates from the console. The following are incorrect coordinates:

* latitude > 90
* latitude < -90
* longitude > 180
* longitude < -180
* -1 < longitude < 1
* -1 < latitude < 1

For another description of this issue see:
https://github.com/tftelkamp/single_chan_pkt_fwd/issues/17

htdvisser, 2017-05-18:
The backend sees this as a "set" location because the GPS location is actually set by the gateway (the altitude is not 0). I agree that it might be good to check for invalid coordinates, so I will add that. Furthermore, I will separate the lat/long from the altitude. This will be done in TheThingsNetwork/gateway-connector-bridge

htdvisser, 2017-05-18:
Resolved by TheThingsNetwork/gateway-connector-bridge@e4968ac1222b743a6c4bf2d22684edd349159123

Not yet deployed, so you might still see this issue

closed 2017-05-18 by htdvisser