TheThingsArchive/ttn #606: Not able to program ThingsNode under Linux

jpmeijers, 2017-05-17:
This is a bug report for Arduino device/ThingsNode.

> What do you want to do?
Program the device.

> What steps did you take?
Arduino IDE, Sparkfun Micro, ATMega 32u4 8MHz, program.

> What went wrong or what is missing?
It did not program.

> What kind of OS/Browser/Gateway are you using? Which version?
Linux Mint, Ubuntu, ...

> Can you fix this yourself and submit a pull request?
Follow the steps here, especially the part about ModemManager:
http://www.arduino.org/learning/getting-started/arduino-ide-on-linux-based-os

johanstokking, 2017-05-17:
Seems like a general serial port on Linux issue, might be in combination with Arduino IDE but might be unrelated as well. Check if the user is in the `dialout` group, which would be a permission issue, although the error message indicates that the port is not found at all.

jpmeijers, 2017-05-17:
The user was in the dialout group from the start. After uninstalling ModemManager it worked once, then not. After a restart of the computer it works again. Seems to keep on working.

johanstokking, 2017-05-17:
If it's specific to the library, which I think it's not, then file the issue in the concerning repository; https://github.com/TheThingsNetwork/arduino-node-lib for example

jpmeijers, 2017-05-17:
No I'm going to open this issue again. The fix is to update the documentation. It's not something we can solve in code.

johanstokking, 2017-05-17:
Replaced by https://github.com/TheThingsNetwork/ttn/issues/606; let's keep this repository clean for routing and issues for non-open source components

closed 2017-05-17 by johanstokking