TheThingsArchive/ttn #613: How to use the api web services of the application and the gateway locally

rabiihlioui, 2017-05-25:
Excuse me but i had to post this issue on the git cause i need a response quikly, i know i should not but i had to, i'm soryy !!!
i have implemented the ttn backend in my localhost, i'm even using my localhost addess as a handler but still have to do authentification phase to the TTN so i can enter and use the system localy, so i had to develop the web services and the authentification service so i no longer have to generate an access key from TTN to work with my local backend, this the hole idea.
But i still don't konw how to connect these services to the local handler (for exemple, to see my applications list, i have developped the service but i don't know how to call it from the backend), any help pls !!!!!!!!!!!!!!!!!!!!!!!!!!!

Thank you

avbentem, 2017-05-25:
Duplicate at https://www.thethingsnetwork.org/forum/t/how-to-use-the-api-web-services-of-the-application-and-the-gateway-locally/7456

johanstokking, 2017-05-25:
Closing this one, will answer on the forum

closed 2017-05-25 by johanstokking