TheThingsArchive/ttn #616: Cannot delete community blog post

h0l0gram, 2017-06-02:
This is a **bug report/feature request** for **the community website**.

I created a post blog in the community page which I later want to delete.
Unfortunately, I can't find a delete button on the post's page or elsewhere to delete the blog post. 
[image]

I would expect a delete button just left of the "publish" button in the post page. 
(similar to the delete features in the console)

gomezjdaniel, 2017-06-02:
It can be deleted from the dashboard page (/dashboard) which only core members have access

closed 2017-06-02 by htdvisser

h0l0gram, 2017-06-02:
Correct. GonzaloCasas pointed that out too.
But I think the author of a post should be able to delete his own posts, no matter what role he/she has within the community.