TheThingsArchive/ttn #618: Adding the metadata in Swagger

SebastienDemarbre, 2017-06-02:
This is a feature request for the console/an integration..

> What do you want to do?
For one of my project, I need some informations about the quality of the transmission. In order words, I need to get the RSSI and SNR. Unfortunately, these values appear in the TTN console but not in the data saved in Swagger. It could be interesting to have a check box to indicate whether we want to save the metadata in Swagger or not. Doing so, we could have more information about the transmission quality withouh requesting the log files from the gateway. 

> What went wrong or what is missing?
The possibility to save the metadata in Swagger to have more information about transmission quality.

johanstokking, 2017-06-06:
Thanks for submitting. This is currently in consideration.

johanstokking, 2017-11-20:
@SebastienDemarbre we're adding this in V3