TheThingsArchive/ttn #619: Add ADR support for other bands than just EU

htdvisser, 2017-06-06:
We currently have ADR support for the EU region, but it shouldn't be too hard to implement this for the other (US, AU, AS, KR) regions as well. @gomezjdaniel: please work with @tftelkamp to implement this. Shouldn't be too complicated, maybe even as simple as copy-pasting the EU settings (https://github.com/TheThingsNetwork/ttn/blob/develop/core/band/band.go#L90) and adapting them for the other regions.

hoonppark, 2017-07-04:
To add ADR support for the Korean region,  I'd like to suggest the following line to be inserted at line 149 of 'band.go' file:
https://github.com/TheThingsNetwork/ttn/blob/develop/core/band/band.go#L149 :

Line # 149: frequencyPlan.ADR = &ADRConfig{MinDataRate: 0, MaxDataRate: 5, MinTXPower: 2, MaxTXPower: 14}

The above line is just a copy of the line # 90 (for EU ADR).

I want someone to verify if the above line has correct parameters for the Korean region ADR.  Please verify the parameters with the LoRaWAN regional spec summary for Korea listed below.

According to the LoRaWAN Spec 1.0.2 Page 40, Data Rates and TXPower for Korea (KR Data Rates and KR TXPower) are as follows:

(1) KR Data Rates -- (from LoRaWAN Spec 1.0.2 Page 40)
----------------------------------------------------------------
Data Rate	           Spreading Factor, Bandwidth
----------------------------------------------------------------
DR0 (   250 bps)	- SF12  BW125 KHz  
DR1 (   440 bps)	- SF11   BW125 KHz
DR2 (   980 bps)	- SF10   BW125 kHz
DR3 (  1760 bps)	- SF9   BW125 KHz
DR4 (  3125 bps)	- SF8   BW125 KHz

DR5 (  5470 bps)	- SF7   BW125 KHz
DR6 ~ 15  : RFU (Reserved for Future Use)
----------------------------------------------------------------

(2) KR TXPower -- (from LoRaWAN Spec 1.0.2 Page 40)
---------------------------------------------------------------
TX0 	  : 20 dbm if supported  (30 dbm = 100 mW)
TX1 	  : 14 dbm (25   mW)
TX2 	  : 10 dbm (10   mW)
TX3 	  :  8 dbm (  6   mW)
TX4 	  :  5 dbm (  3.2 mW)
TX5 	  :  2 dbm (  1.6 mW)
TX6	  :  0 dbm (
TX7 ~ 15  : RFU (Reserved for Future Use) 
---------------------------------------------------------------

avbentem, 2017-12-30:
For future reference, US and AU have meanwhile been implemented as well:

<https://github.com/TheThingsNetwork/ttn/releases/tag/v2.9.0> (October 2017):

> The 2.9.0 release contains the following new features and improvements:
>
> - Added ADR support for US and AU regions
> - ...

htdvisser, 2019-02-18:
With #755 merged, I think we can close this one

closed 2019-02-18 by htdvisser