TheThingsArchive/ttn #624: Can not edit wiki

jpmeijers, 2017-06-09:
This is a bug report for the wiki.

> What do you want to do?

Edit an article.

> What steps did you take?

Go to the article, and click on edit at the top right. Add modifications and click on save.

> What went wrong or what is missing?

I am redirected to a "create article" page.

> Do you have Screenshots?

[image]

[image]

> What kind of OS/Browser/Gateway are you using? Which version?

Chromium Version 56.0.2924.84 Built on Ubuntu 14.04, running on Raspbian 8.0

> Can you fix this yourself and submit a pull request?

No. But a workaround is to edit the article on github, but you need special permissions for that.

dorianamouroux, 2017-07-05:
We can now edit on the website

johanstokking, 2017-07-17:
So this is resolved?

closed 2017-07-17 by johanstokking