TheThingsArchive/ttn #627: Gateway location not on NOC but set on Console

jpmeijers, 2017-06-17:
This is a bug report for the bridge/the NOC.

> What do you want to do?

Get the location of a gateway.

> What steps did you take?

See and compare:
http://noc.thethingsnetwork.org:8085/api/v2/gateways/eui-5ccf7fffff2bb020
https://account.thethingsnetwork.org/gateways/eui-5ccf7fffff2bb020

> What are the IDs and EUIs of your Device/Gateway? (if applicable)

eui-5ccf7fffff2bb020

> What I expected

The NOC should fall back to the location from the account server if the gateway does not send a location. In this case it did not fall back.

htdvisser, 2017-06-20:
None of our routers has received any "status" message from this particular gateway, hence there is no location set in the NOC. Furthermore, it has not been seen since last week. Please check the gateway connectivity.

jpmeijers, 2017-06-20:
I have seen TTN Mapper packets via this gateway on 2017-06-16. At this stage I did not see any coordinates for this gateway on the NOC. A day or two later when going through the data, I saw the coordinates for this gateway on the account server. 

The only thing that explains this behaviour then is if the gateway owner set his gateway location on the console only after the gateway went offline.

closed 2017-06-20 by jpmeijers