TheThingsArchive/ttn #646: Estimated airtime in TTN Console's gateway and device traffic are different

avbentem, 2017-07-02:
This is a **bug report** for **the console**.

> What do you want to do?

Get the estimated airtime of a packet in TTN Console.

> What steps did you take?

Sent a 21 bytes packet (8 bytes application payload) on SF7. For this, the LoRaWAN air time calculation sheet at https://www.thethingsnetwork.org/forum/t/spreadsheet-for-lora-airtime-calculation/1190 shows 56.576 ms airtime. Also, the gateway's traffic page shows: "airtime (ms) = 56.6 ms".

(When repeating with 17 bytes total, being 4 bytes application payload, the sheet calculates 51.456 ms, and the gateway then shows 51.5 ms too.)

> What went wrong or what is missing?

For 21 bytes, the device's traffic page shows "Estimated Airtime = **51**.456 ms" instead of **56**.5 ms like the gateway does. 

**It seems 2 bytes too few are taken into account for the estimation?** For 19 bytes, the calculation sheet indeed shows 51.456 ms.

(When repeating with 17 bytes, the device traffic shows 46.336 ms, while the gateway shows 51.5 ms. Again: like 2 bytes fewer were used for the calculations; for 15 bytes the sheet indeed shows 46.336.)

> Do you have Screenshots?

Gateway:

[image]

Device:

[image]

...after scrolling down:

[image]

Airtime calculator:

[image]

> What kind of OS/Browser/Gateway are you using? Which version?

Mac OS X, Chrome, single-channel test gateways.

> What are the IDs and EUIs of your Device/Gateway? (if applicable)

- Gateways: eui-b827ebffff5fe05c and eui-5ccf7ff42f1970ec
- Device: arjanvanb-bb-2/EUI 0004A30B001B1874

> What do your configuration files look like?

n/a

> What do your log files look like?

n/a

> Can you fix this yourself and submit a pull request?

Maybe, if you open source the code ;-)

romeovs, 2017-07-03:
@jpmeijers Can you have a look why this is wrong?

johanstokking, 2017-07-17:
Isn't this fixed @Ragnn @romeovs ?

romeovs, 2017-07-17:
Not yet. I'll work on it soon. @Ragnn added the backend part, but the front-end needs to be updated to reflect that.

romeovs, 2017-10-09:
This should be fixed by fe151b7f53. Can you confirm @avbentem?

htdvisser, 2017-11-03:
@romeovs: This issue is still not resolved (as far as I can see on the PCN console).

Take as an example the `office-app`. Our sensor at HQ should show an airtime of `51.456`. The gateway view shows the correct value, but the Data view doesn't. 

Today I started with the implementation of #682, which adds the airtime to uplink (and downlink event) metadata. When deployed, that value can be displayed in the console.

romeovs, 2017-11-06:
The estimated airtime shown in the console is calculated by the `github.com/TheThingsNetwork/ttn/utils/toa` package. Could it be that the parameters are interpreted wrongfully?

htdvisser, 2017-11-09:
Both values are calculated by the console. One is correct, another isn't.

[image]

romeovs, 2017-11-09:
Coule it be that the console gives the wrong parameters to the `toa` package? https://github.com/TheThingsIndustries/console/blob/master/server/api/events/app_subscription.go#L302-L305

Difference between PHY payload and application payload could be the cause?

htdvisser, 2017-11-09:
Likely. In that case it can be resolved when https://github.com/TheThingsNetwork/ttn/commit/0f795e39c6f00b663f70212595de09cf4f2c92a4 is live

sigmaroot, 2018-09-11:
Is this still unsolved? I'm seeing some differences and I think both, gateway and application estimated airtimes are wrong.

Sending 38 byte of app payload, resulting in 51 byte payload in the gateway overview:

[image]

In the application:

[image]

And this is calculated with the Semtech LoRa Model Calculator Tool (51 bytes):

[image]

And with 38 bytes:

[image]

So, we do have 4 different airtimes, none matching the TTN airtimes.

Which one is the correct one?

kizniche, 2019-03-24:
I recently had The Things Network stop the movement of all my data from my gateway to my apps due to going over the 30-second daily limit (not just the offending app/device, but ***all*** of my apps, even the ones not exceeding 30 seconds/day). This was due to using the App airtime estimate. Using the gateway airtime, data began appearing in my apps again. I see reporting incorrect airtime (or calculation) as a significant bug.

Additionally, is it expected behavior for data from all apps to cease operating when one device exceeds the daily transmission limit?

htdvisser, 2019-03-25:
Hi @kizniche, the 30-second daily limit is not enforced; we don't even have any code to do that, so I can with 100% certainty say that the lack of traffic was not caused by this.

kizniche, 2019-03-25:
Thank you for the clarification. Perhaps it's just a coincidence they all stopped. I'll have to investigate further.

closed 2021-06-03 by htdvisser