TheThingsArchive/ttn #657: Use github.com/goreleaser/goreleaser

rvolosatovs, 2017-07-16:
This is a **feature request** for **the backend/ttnctl/the console/the NOC/integrations**.
Let's use https://github.com/goreleaser/goreleaser to automate, standardize and ease releasing process. In the context of `ttnctl` and other public applications, this also allows us to automatically update the Homebrew package and here should be a way to extend it to add AUR/Debian/NixOS etc. support. I would be willing to contibute to that, also in my own free time.
refs #255 #648
What do you think?
@htdvisser @romeovs @johanstokking @dotpy3

htdvisser, 2017-07-17:
Absolutely. I've been using it in https://github.com/htdvisser/squatt and it works very well. Also works great with Travis. 

Let's wait until we split `ttnctl` into a separate repository though

htdvisser, 2017-08-21:
We decided to keep working with our `Makefile`-based build. Closing the issue

closed 2017-08-21 by htdvisser