TheThingsArchive/ttn #662: Migrate to github.com/TheThingsNetwork/api

htdvisser, 2017-07-19:
We currently use `github.com/TheThingsNetwork/ttn/api`, but this is being moved to `github.com/TheThingsNetwork/api`.

In the process we are making some changes to the generated Go code in the new `api` package:

- `Id` is renamed to `ID`
- `Eui` is renamed to `EUI`
- Some fields are now marked as non-nullable

Please report any other improvements in https://github.com/TheThingsNetwork/api/issues

rvolosatovs, 2017-07-19:
As far as I understand, packages like `api/stats` and `api/ratelimit` are not moved, or are they? Please specify concrete packages being moved

htdvisser, 2017-07-19:
The packages that contain `.proto` files

```
.
├── broker
├── discovery
├── gateway
├── handler
├── logfields
├── monitor
├── networkserver
├── protocol
│   └── lorawan
├── router
└── trace
```

rvolosatovs, 2017-07-19:
what should happen with the Clients, i.e. `monitor.Client` ?

htdvisser, 2017-07-19:
There are a number of things that will have to be moved to the new API package:

- Convenience functions (for example `*DeduplicatedUplinkMessage.InitResponseTemplate()`)
- Some implementations (for example in `trace`)

There are a number of things that will need a new place somewhere else:

- `Message`/`Payload` (un)marshaling
- Reference Servers
- Stream clients
- Stream servers
- gRPC Client Pool
- Ratelimiting
- `stats`

Or have to be removed completely:

- Discovery client "overlay"
- Handler Manager client "overlay"

rvolosatovs, 2017-07-19:
I will take care of that also then

htdvisser, 2017-07-19:
Great! Nice to finally clean this mess up 😄

commit 6cd4035d (2017-07-20): migrate to github.com/TheThingsNetwork/api

Closes #662
commit 2810ff9c (2017-07-21): migrate to github.com/TheThingsNetwork/api

Closes #662
commit f623a6a1 (2017-07-24): migrate to github.com/TheThingsNetwork/api

Closes #662
PR #665 (2017-07-25): Migrate to github.com/TheThingsNetwork/api
Closes #662
closed 2017-07-25 by htdvisser