TheThingsArchive/ttn #668: when port is nil will cause a MIC VALID

RongAtRabbit, 2017-08-18:
This is a  MIC VALID bug report  for the backend

> What do you want to do?
I want to receive a ack message from lora module
> What steps did you take?
receiving the ack
> What went wrong or what is missing?
The Message with port==nil and len(frmpayload) == 0. The backend report  "Invalid MIC"
> Do you have Screenshots?
no
> What kind of OS/Browser/Gateway are you using? Which version?
nil
> What are the IDs and EUIs of your Device/Gateway? (if applicable)
nil
> What do your configuration files look like?

> What do your log files look like?

> Can you fix this yourself and submit a pull request?

RongAtRabbit, 2017-08-18:
I think the reason is in MACPayloadFromPayload;
		if payload.FPort != nil { ==== >when payload.FPort== nil,it will init the mac.FPort = 0
			mac.FPort = int32(*payload.FPort)
		}
but in function : 
// Payload converts the MACPayload to a lorawan.Payload
func (msg *Message_MacPayload) Payload() lorawan.Payload {
	if m.FPort >= 0 { ====>this step will give port a value 0
		fPort := uint8(m.FPort)
		mac.FPort = &fPort
	}
above change the port nil to value 0
MIC is calculated by  port nil. but will validate by port value 0. so it cause the "Invalid MIC"

htdvisser, 2017-08-21:
`MACPayloadFromPayload` will indeed output an FPort of `0` for both situations (FPort=0 and FPort=nil). However, the `Payload()` will only set the FPort if it is greater than 0 or set it to 0 if there is a FRMPayload. In other cases it will remain `nil`:

```go
if m.FPort > 0 {
	fPort := uint8(m.FPort)
	mac.FPort = &fPort
}
if len(m.FRMPayload) != 0 {
	mac.FRMPayload = []lorawan.Payload{&lorawan.DataPayload{Bytes: m.FRMPayload}}
	if mac.FPort == nil {
		fPort := uint8(0)
		mac.FPort = &fPort
	}
}
```

Could you give a little more information about the device you are using for this? It would be very helpful to have the

- Raw payload (as received by the gateway) of the problematic frame
- NwkSKey that was used at the time of the frame

closed 2017-10-11 by htdvisser