TheThingsArchive/ttn #677: Coordinates on NOC wrong, even if from REGISTRY

jpmeijers, 2017-09-25:
This is a bug report for the NOC.

> What do you want to do?

Get the location of a specific gateway.

> What steps did you take?

Look the location up on: http://noc.thethingsnetwork.org:8085/api/v2/gateways/eui-aa555a0000088213
I get: `"latitude":51.7148,"longitude":5.35617,"altitude":9,"source":"REGISTRY"`

Look the location up in the "registry": https://account.thethingsnetwork.org/gateways/eui-aa555a0000088213
I get: `"altitude":65,"latitude":52.21176,"longitude":5.96243`

> What went wrong or what is missing?

The coordinates on the NOC is not the same as in the registry, even though that is indicated as the source.

htdvisser, 2017-09-27:
If I look up the gateway with `ttnctl`, I get the following:

```
❯ ttnctl gateways status eui-aa555a0000088213

  INFO Received status                          GatewayID=eui-aa555a0000088213

           Last seen: 2017-09-27 09:28:20.032534773 +0200 CEST
           Timestamp: 0
       Reported time: 2017-09-25 22:23:54 +0200 CEST
         Description: lora node
            Platform: IMST + Rpi
      Frequency Plan: EU_863_870
              Bridge: br-eu1 2.x.x-bdf5610cd762522020e16b275b9908a01151a7bb (2017-07-03T13:12:07Z) PacketForwarder Backend
            Location: (51.714802, 5.356170; source unknown)
```

So it looks like this location is actually sent by the Gateway, not taken from the registry. 

@rvolosatovs: could you check why the NOC reports the location as REGISTRY?

rvolosatovs, 2017-09-27:
The issue is because the "source" of gps is `UNKNOWN`, the source value is discarded in that case, as it's assumed to be not set.
Apparently gateway has once reported it's gps source as `REGISTRY`, that value got cached and had been used ever since.
I removed the `source` value from cache for the gateway, now it's not reported.

@htdvisser do we want to change this behaviour? In my opinion everything works as expected - wrongly cofigured components show wrong data

htdvisser, 2017-09-27:
`UNKNOWN` is a valid value for the LocationSource enum. If a gateway's location is reported as `UNKNOWN` it should be stored like that, not discarded. 

> wrongly cofigured components show wrong data

I agree, but then the "wrong" location should match the "wrong" location source. Now the NOC returns a "wrong" location and blames the "REGISTRY" source

closed 2017-10-05 by htdvisser