TheThingsArchive/ttn #681: Changing an application's handler no longer warns that all devices will be lost

ama9910, 2017-10-13:
> What do you want to do?
Improve communication to users

> What steps did you take?
Create application, register device, change to different hander, note that existing device no longer exists

> What went wrong or what is missing?
Warn user that devices will be deleted

> Can you fix this yourself and submit a pull request?
Unfortunately not

htdvisser, 2017-10-13:
I assume this issue is related to the console?

ama9910, 2017-10-14:
Yes, Console

Andrew

romeovs, 2017-10-16:
Do you see errors in the developer console?
I've tested here on a couple of browsers and everything seems to work as expected.

ama9910, 2017-10-16:
> Do you see errors in the developer console?

There are no errors associated with this issue.

> everything seems to work as expected

How is the user notified that devices will be lost? All I see is the following message "Are you sure you want to change the handler registration for application .....? Confirm by typing the name of the application:"

romeovs, 2017-10-16:
I see what you mean. I thought the issue was that you did not see this message. But the issue is that the message is not descriptive enough.

I'll fix it soon!

ama9910, 2017-10-20:
Thanks! FYI a previous console version would notify the user that devices configurations would be lost. I don't recall which version or what the message said.

htdvisser, 2017-11-03:
[image]

closed 2017-11-03 by htdvisser