TheThingsArchive/ttn #682: Add airtime to message metadata

gonzalocasas, 2017-11-03:
This is a **feature request** for **the backend**.

> What do you want to do?

I would like to be able to analyze and understand the air time used by each device over time.

> What is missing?

The airtime is currently available in the TTN console but only as an *on-the-fly* calculation ([airtime calculator](https://docs.google.com/spreadsheets/d/1voGAtQAjC1qBmaVuP1ApNKs1ekgUjavHuVQIXyYSvNc/edit?usp=sharing)). However, this value is not accessible via any API. It would be very useful to have this value (even if it is an estimate), integrated into the payload of the messages, as part of the metadata.

gonzalocasas, 2017-11-03:
Implementation might be linked to #646.

htdvisser, 2017-11-03:
I like the idea. Then at least we don't have to worry about inconsistent implementations anymore.

gonzalocasas, 2017-11-03:
/cc @maveonair

commit fa0e9cdb (2017-12-18): Add airtime to uplink metadata and downlink sent event

Fixes #682
commit 997ecc3e (2017-12-22): Add airtime to uplink metadata and downlink sent event

Fixes #682
closed 2017-12-22 by htdvisser