TheThingsArchive/ttn #686: Downlink to LoRa devices

mazareth, 2017-12-01:
Hi everybody.
This is a question for an integration.

> What do you want to do?
use a watteco temp/hum sensor

> What steps did you take?
Add the device on thethingnetwork, thats work

> What went wrong or what is missing?
1 - I don't understand why thethingnetwork send a message with empty payload after the temp or hum frame. Where this message come from?
2 - I want to send configuration message to my device but i don't understand how to do.
I try in the dowlink tab... that appear in application data but nothing more...
Do you have explanation?

> Do you have Screenshots?
[image]

[image]

[image]

> What kind of OS/Browser/Gateway are you using? Which version?
packet_forwarder on pi with picoGW
drive thethingnetwork by chrome on windows 7

> What are the IDs and EUIs of your Device/Gateway? (if applicable)
gateway : eui-414456414e54454e

johanstokking, 2017-12-01:
@mazareth thanks for reporting.

1. Payload on FPort 0 are MAC commands, these are sent by the network server as per LoRaWAN specification;
2. Seems like you do the configuration right. You can send the downlink by pressing the button. You will see it appear scheduled in the Application Data, and then sent off the next time there's an opportunity to send

mazareth, 2017-12-01:
Hi Johan,
Thanks for 1.

For 2 - Do you know where the opportunity to send is calculated?
What's the Fport change?
What's the confirmed checkbox change?

johanstokking, 2017-12-01:
@mazareth yes, on class A the network has an opportunity to send only directly after an end device's uplink transmission.

I'm closing this issue as I think it is resolved.

Did you know that there's a #support channel on our Slack? Also, there's the public forum.

mazareth, 2017-12-01:
@johanstokking  no i don't know where i can't ask question.
Could you give me a link?

And how can i change the class of the network?

johanstokking, 2017-12-01:
Class C is experimental in TTN Stack V2 (this version), but class B and C will be generally available in TTN Stack V3, launched 2018 Q1.

closed 2017-12-01 by johanstokking

htdvisser, 2018-07-16:
See [this post on the forum](https://www.thethingsnetwork.org/forum/t/ttn-network-stack-v3-central-1/14991/6?u=htdvisser)