TheThingsArchive/ttn #697: Remove scheduled downlink

sukoy, 2018-02-23:
Hi!

There is a way to remove a scheduled downlink from queue?
Our devices use OTAA and after reset cnt goes to 0, is there a way to remove previous scheduled downlink? to avoid twice received message.

Thanks

johanstokking, 2018-02-23:
@sukoy atm there's no way to clear the queue, apart from replacing it with a new value. This will be added in V3.

sukoy, 2018-02-23:
ok Thanks

closed 2018-10-05 by htdvisser

haorfani, 2019-08-26:
I am still having trouble with this issue. If I schedule a downlink message from an application, it is sent forever. Is there a way to stop it?

johanstokking, 2019-08-27:
@haorfani that should not happen. Are you sending confirmed downlink and is the downlink not confirmed by the end device, by chance? Because in that case it works as expected.

haorfani, 2019-08-27:
By this you mean that the end device should send a ACK packet back to the GW right?

haorfani, 2019-08-27:
OK I fixed it, it was my own bug after all. Thanks for the response though!

pradeeka7, 2020-06-21:
I have the same issue with my RAKwireless 7205 tracker. This happened from the second day. First day it worked well. I haven't yet restarted the tracker.

I'm sending UNCONFIRMED messages. I can't send confirmed messages because the latest firmware has a bug (I guess).

[image]

jpmeijers, 2020-06-21:
Downlinks on port 0 are mac commands. Most likely what you are seeing is the ADR downlinks telling your device to go to a faster datarate and/or lower output power.