TheThingsArchive/ttn #702: DevEUI not valid by device registration on-join

neolinux, 2018-03-22:
This is a **bug report/feature request/question/...** for **the website/the backend/ttnctl/the console/the NOC/an integration/...**.

> What do you want to do?
Activate device registration on-join on one Application

> What steps did you take?
Downloaded the ttn-cli x64 for Windows
Launched this command:
./ttnctl-windows-amd64.exe device register on-join demo C7E94AC4B4AED1E8D6F78E**********

> What went wrong or what is missing?
After that Command i became this error back:

  INFO Using Application                        AppEUI=70B3D57ED000B101 AppID=pm-auto
  WARN On-join registration does not follow LoRaWAN best practices. Unless you implement additional security
  WARN measures, it may corrupt your entire application if even one of your devices gets compromised.
  INFO Discovering Handler...                   Handler=ttn-handler-us-west
  INFO Connecting with Handler...               Handler=us-west.thethings.network:1904
  INFO Application does not have a RegisterOnJoinAccessKey
  INFO Looking for existing key...
  INFO Using existing key...
 FATAL Could not register Device                error=Could not set device on Handler: Invalid Device: Invalid Device: Invalid LoRaWANDevice: DevEUI not valid: can not be empty source=Wrap: /go/src/github.com/TheThingsNetwork/ttn/utils/errors/errors.go:222

johanstokking, 2018-03-28:
Thanks! We'll have a look at this. This could be regression when validating values.

neolinux, 2018-03-29:
Thank You,

i will wait for a Solution.

johanstokking, 2018-03-29:
@dotpy3 will look into it, thanks.

jpmeijers, 2018-03-30:
I am also affected by this bug.

htdvisser, 2018-10-05:
This will be resolved in the next release. We were a bit too strict with our validations. Please do note that on-join registration is typically not a great idea, and we may stop supporting it in v3.

closed 2018-10-05 by htdvisser