TheThingsArchive/ttn #703: Ghost messages in consle

j4zzcat, 2018-03-25:
Hi,

I'm seeing ghost messages in the console. 
I don't have any application registered, I'm merely testing with the things node. 

Clicking on the things node button will result in two messages in the console. One is for real, the other is a ghost. After power-cycling the things node, the ghost messages disappear. 

Screenshot, notice the ghost messages with count > 28800.
[image]

Those ghost messages always seems to have the following in their log:
```
cmd:link-adr
error:too many failed ADR requests
```

And the payload looks like this:
```
{
  "gw_id": "j4zzcat-ttngw-1",
  "payload": "YCwdASYg1nCGCCkK",
  "f_cnt": 28886,
  "lora": {
    "spreading_factor": 9,
    "bandwidth": 125,
    "air_time": 144384000
  },
  "coding_rate": "4/5",
  "timestamp": "2018-03-25T12:32:53.598Z",
  "dev_addr": "26011D2C",
  "frequency": 869525000
}
```

KrishnaIyer, 2018-03-26:
@j4zzcat : Could you please comment on why you closed this issue? it's useful if we look back into this issue.

j4zzcat, 2018-03-26:
@krishnalyer I’m not sure this was really a problem. I want to further explore the issue before reporting.

htdvisser, 2018-04-03:
I see the original message was edited with more information.

From what I can see, the situation is as follows:

- The device is using ADR
- The device previously rejected multiple ADR commands
- The network server is attempting to optimize the data rate (on the uplink flow)
- The network server is rejecting its own data rate optimization (on the downlink flow) because of too many failed requests

Files to look at:

- https://github.com/TheThingsNetwork/ttn/blob/develop/core/networkserver/adr.go
- https://github.com/TheThingsNetwork/ttn/blob/develop/core/networkserver/uplink_mac.go