TheThingsArchive/ttn #707: console/account server: Changing e-mailadress *appears* to not work

matthijskooijman, 2018-04-19:
This is a **bug report** for **the console**.

> What do you want to do?
I changed my e-mail-address on the console (or account server?).

> What steps did you take?
I changed my address at https://account.thethingsnetwork.org/users/profile

> What went wrong or what is missing?
A message " Profile successfully updated! " is shown, but then the original, unchanged e-mailadress is still shown below, giving the impression that the address was not changed. A later realized that it sent an e-mail verification message, and only shows the new email address is not shown until it is verified.

Furthermore, when I (accidentally) changed the e-mail address *back* to the original (before verifying the new one), it showed an additional message: " Check you email for a link to update the email address of your account. ".

So the actual bug I'm reporting is that the additional message is not shown when changing your e-mailaddress, only when changing it back.

> What kind of OS/Browser/Gateway are you using? Which version?
Firefox 52 on Linux Debian stretch.

johanstokking, 2018-04-20:
Indeed, the change is effective from the moment the new email address is validated.

It should indeed mention this explicitly, that's wrong. We don't address this in V2 anymore.

Thanks for reporting anyway!

htdvisser, 2021-06-03:
We won't fix this for the V2 Account Server, but we'll keep it in mind for The Things ID

cc: @bafonins