TheThingsArchive/ttn #709: Some issues in docker-compose.yml?

Avasz, 2018-04-23:
This is a **bug report/question/...** for **tttnstack**.

> What do you want to do?
run ttn stack via docker-compose.yml

> What steps did you take?
cloned repository and ran `docker-compose up`

> What went wrong or what is missing?
there seems to be some problem with environment variables for "rabbitmq" and "bridge" services.

> Do you have Screenshots?
no.

> What kind of OS/Browser/Gateway are you using? Which version?
Debian 9

> Can you fix this yourself and submit a pull request?
Maybe, but please confirm my doubts, as it worked when I did it. At least `docker-compose up` was successful. haven't checked the working functionality of the whole stack.

1. rabbitmq searches for "ca.cert" in `/var/lib/rabbitmq/tls/ca.cert` which should work fine if not in docker because it seems to be symlinked to `../discovery/server.crt`. but for docker it doesn't work. Adding the following line under "volume" works:   
`./env/discovery/server.cert:/var/lib/rabbitmq/tls/ca.cert`  
There might be some better method too. 

2. It seems that by default the `bridge` container searches for `redis` in localhost:6379. Adding the following line under `environment` of bridge service seems to work:  
`BRIDGE_REDIS_ADDRESS: redis:6379`  

Please check it and let know if what I did are correct.

Thanks

KrishnaIyer, 2018-04-23:
> BRIDGE_REDIS_ADDRESS: redis:6379

This is correct. 

Please use the updated tutorial in the link below and let us know how things go:
https://www.thethingsnetwork.org/article/deploying-a-private-routing-environment-with-docker-compose

Avasz, 2018-04-23:
Ah nice. 
I didn't think there would be such a detailed guide for docker-compose up, so didn't bothered to search, my bad.

Everything is writted in detail there. 
Thank You!

closed 2018-04-23 by KrishnaIyer