TheThingsArchive/ttn #711: Add Band Plan information to pop-up dialog for gateways on TTN Map

ama9910, 2018-05-22:
This is a feature request for the website.

> What do you want to do?
In Australia we are fortunate/unfortunate enough to be able to use both AU915 and AS923 band plans. When looking at the TTN map, it is not obvious to a user which band plan a gateway is using resulting in frustration and wasted time trying to make a connection on the wrong band plan.

> What went wrong or what is missing?
It would be great if the pop-up dialog containing information about the gateway also includes the band plan information. eg..

Name: kpmg-iotaa-meshed-barangaroo
Owners: maj
Altitude (m): 170
Placement: outdoor
Brand: MultiTech
Model: Conduit AEP
Antenna model: Not specified
**Band Plan: AU915**

If space is limited, I suggest removing "Antenna Model".

> What are the IDs and EUIs of your Device/Gateway? (if applicable)
For example, macquarie_uni-meshed-1 is AS923, kpmg-iotaa-meshed-barangaroo is AU915

johanstokking, 2018-05-22:
Thanks, good suggestion, especially relevant in Australia.

We'll take this into account in V3 because we'll have the information at hand without changes.

htdvisser, 2021-06-03:
This is not going to be added in V2 anymore, but the mapping API of Packet Broker [includes the frequency plan of the gateway](https://github.com/packetbroker/api/blob/master/mapping/v2/openapi.tmpl.json#L418).

closed 2021-06-03 by htdvisser