TheThingsArchive/ttn #716: Add frequency plan for 915-917

j4zzcat, 2018-06-05:
Hello,

A plan for Israel (IL) is missing from the list. 
Here are the details: 
915-917MHz
In 916.3MHz allowed 100mA 20dbm

nestorayuso, 2018-06-05:
https://www.thethingsnetwork.org/community/telaviv/post/915mhz

there is not a valid regional parameters por israel defined by the lora alliance

htdvisser, 2018-06-05:
According to the current (June 2018) [LoRaWAN Regional Parameters](https://lora-alliance.org/sites/default/files/2018-04/lorawantm_regional_parameters_v1.1rb_-_final.pdf), the only frequency band that is specified for Israel is EU433.

We can not just start creating and supporting unspecified frequency plans, so for the time being, your only option is EU433. You may want to contact the LoRa Alliance to see if you can help them in any way with the specifications for Israel.

closed 2018-06-05 by htdvisser

abard, 2018-10-23:
Please excuse me @htdvisser  for posting in a close issue. 
However, I didn't understand your answer. 
915 - 917 MHz range is clearly specified for use in  Israel according to the (June 2018) LoRaWAN Regional Parameters. What else should the LoRaWAN Regional Parameters include so that you can create and support this document?  
I have attached a screenshot of the relevant part in the document.  
[image]

htdvisser, 2018-10-23:
This indicates that 433.05-434.79 is available spectrum and the EU433 specification can be used. 915-917 is available spectrum as well, but there is no specification that can be used. Nothing stops you from deploying gateways that listen on those frequencies, implementing devices that send LoRaWAN messages on those frequencies, and configuring your gateways to forward to TTN. Unfortunately though, TTN won't be able to reply to those messages, because there is no specification that tells us how.

abard, 2018-10-23:
Thank you @htdvisser for your swift response!
Can you offer me a possible setup with TTN? How should I register my GW's in TTN? on what frequency plan and how will it affects the activation setup for the registered devices?

htdvisser, 2018-10-24:
I'm afraid I can't. TTN's official recommendation is to wait until the LoRa Alliance publishes a regional parameters specification that is suitable for Israel's 915-917 band. Until then you could experiment with the 433MHz band (but you will need different hardware for that).

I do know that the LoRa Alliance has been working on a "middle east" regional specification, but I can't share any details about that yet.

I'm very sorry that I can't be of more help, this is out of TTN's control, so we'll just have to be patient.

avbentem, 2019-01-11:
Just for future reference, from https://github.com/TheThingsNetwork/docs/pull/232#issuecomment-447736529:

> We're not planning to support EU433 officially.

But maybe that only applies to usage in Europe, not in, e.g., Israel.