TheThingsArchive/ttn #718: Dragino - TTN connection failed

AKROF, 2018-06-12:
This is a **bug report/feature request/question/...** for **the website/the backend/ttnctl/the console/the NOC/an integration/...**.

> Connect Dragino LG01-P to TTN

> I followed Dragino wiki and registered my Gateway in TTn and also made the required changes on dragino page

> The TTN console doesnot show the dragino gateway as connected

> Do you have Screenshots?
[image]

I configured the Gateway files with all the details as given in the wiki
http://wiki.dragino.com/index.php?title=Connect_to_TTN

Can some one suggest what went wrong

KrishnaIyer, 2018-06-22:
Hi @AKROF 

Have you checked the Network Troubleshooting steps described here:
http://wiki.dragino.com/index.php?title=Connect_to_TTN#Why_TTN_gateway_shows_status_.22not_connected.22.3F

Without any log/error messages, it is difficult for us to provide you suggestions.

**TIP:** You can ask this question in the `# gateways` channel on the `The Things Network Community Slack`.  
We have some amazing active community members who might offer you suggestions as they've likely had similar issues.

AmPalRed, 2018-06-26:
Hey Hi, 
Thank you for the reply. I was able to set up the gate way.
I  reset the whole thing again from start.

closed 2018-06-26 by AmPalRed

enthutech, 2019-05-20:
You can visit https://www.youtube.com/watch?v=BqUpkWl_EsQ&list=PLDQtDl_T-tm3ehh4iCMY--CWnrDXh1nDB&index=3 for video tutorials on dragino product working