TheThingsArchive/ttn #726: Support for custom frequency plan

j4zzcat, 2018-07-04:
This is a feature request.
Allow the user to define a custom frequency plan for a gateway.

KrishnaIyer, 2018-07-04:
Oops. 

@j4zzcat: Can you please use our template while creating issues?

What exactly do you mean by `custom frequency plan`? can you provide some examples?

htdvisser, 2018-07-05:
This will likely not be possible on the public community network, but with our v3 stack, some of this should be possible in a private network.

Things that we will most probably **never support**:

- Change the supported SFs of US915/AU915/CN470 channels - This would make the network non-compliant with LoRaWAN
- Change the fixed join channels of EU868/CN779/AS923/... - This would make the network non-compliant with LoRaWAN

j4zzcat, 2018-07-05:
Indeed, my intention was that this feature be used in private networks.

htdvisser, 2019-02-18:
In v3 it is now possible to configure a network server with a custom frequency plan repository (typically a fork of https://github.com/TheThingsNetwork/lorawan-frequency-plans).

closed 2019-02-18 by htdvisser