TheThingsArchive/ttn #728: Deploying a Private Routing Environment:  No communication between Gateway and Router/bridge

AnilkumarKomanduru, 2018-07-13:
I have installed private TTN network and gone through articles - https://www.thethingsnetwork.org/article/deploying-a-private-routing-environment-with-docker-compose
https://www.thethingsnetwork.org/article/setting-up-a-private-routing-environment. 

I have run 'docker-compose up -d ' and all the services are started successfully.
Now, i have changed the server_address in gateway( which has ttn software) to hostname i.e."192.168.0.163" and left port as it is 1700

when i try to analyze packets are transmitted from gateway on 1700 port but no respose from the server.
Found that 'destination unreachable' from the wireshark even though the port is opened.

I have registered the gateway using 'ttnctl' and configured as Router - mynetwork-router.

I would like to know where can i see the server status ? 

ttnctl.yml:

         config file: /home/xx/.ttnctl.yml
            data dir: /home/xx/.ttnctl

        mqtt-address: localhost:1883
           router-id: mynetwork-router
       mqtt-password: 
          handler-id: mynetwork-handler
       gateway-token: eyJhbGciOiJSUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJ0dG4tYWNjb3VudC12MiIsInN1YiI6ImRldiIsInR5cGUiOiJnYXRld2F5IiwiaWF0IjoxNDgyNDIxMTEyfQ.obhobeREK9bOpi-YO5lZ8rpW4CkXZUSrRBRIjbFThhvAsj_IjkFmCovIVLsGlaDVEKciZmXmWnY-6ZEgUEu6H6_GG4AD6HNHXnT0o0XSPgf5_Bc6dpzuI5FCEpcELihpBMaW3vPUt29NecLo4LvZGAuOllUYKHsZi34GYnR6PFlOgi40drN_iU_8aMCxFxm6ki83QlcyHEmDAh5GAGIym0qnUDh5_L1VE_upmoR72j8_l5lSuUA2_w8CH5_Z9CrXlTKQ2XQXsQXprkhbmOKKC8rfbTjRsB_nxObu0qcTWLH9tMd4KGFkJ20mdMw38fg2Vt7eLrkU1R1kl6a65eo6LZi0JvRSsboVZFWLwI02Azkwsm903K5n1r25Wq2oiwPJpNq5vsYLdYlb-WdAPsEDnfQGLPaqxd5we8tDcHsF4C1JHTwLsKy2Sqj8WNVmLgXiFER0DNfISDgS5SYdOxd9dUf5lTlIYdJU6aG1yYLSEhq80QOcdhCqNMVu1uRIucn_BhHbKo_LCMmD7TGppaXcQ2tCL3qHQaW8GCoun_UPo4C67LIMYUMfwd_h6CaykzlZvDlLa64ZiQ3XPmMcT_gVT7MJS2jGPbtJmcLHAVa5NZLv2d6WZfutPAocl3bYrY-sQmaSwJrzakIb2D-DNsg0qBJAZcm2o021By8U4bKAAFQ
   discovery-address: localhost:1900
              app-id: 
             app-eui: 
      allow-insecure: false
       mqtt-username: 
         auth-server: https://account.thethingsnetwork.org
                data: /home/xx/.ttnctl
              config: /home/xx/.ttnctl.yml
          gateway-id: dev

local_conf.json file:

{
  "gateway_conf": {
      "gateway_ID": "0000024b080501b0",
      "serv_port_up": 1700,
      "serv_port_down": 1700,
      "server_address": "192.168.0.163",
      "forward_crc_valid": true,
      "forward_crc_error": false,
      "forward_crc_disabled": true,
      "gps": true
  }
}

johanstokking, 2018-07-13:
Community support for setting up private networks is provided through the forum and Slack, see also the bottom of the articles that you reference;

>If you have questions or remarks after following this guide, feel free use the [forum](https://www.thethingsnetwork.org/forum) or the `#private-backend` channels on Slack. This is a community-supported guide, so please help each other out.

Cross-posting on the forum, here and in different Slack channels is counter-productive for you and for the community.

closed 2018-07-13 by johanstokking

AnilkumarKomanduru, 2018-07-13:
Thanks @johanstokking. I don't have option to post a query in the forum?
I couldn't find the 'reply' tab with my account where as it is working fine with others account.