TheThingsArchive/ttn #729: Claim orphaned gateway registration

jpmeijers, 2018-07-13:
This is a feature request for ttnctl/the console.

# What do you want to do?

Claim ownership of a gateway that is in my possession, but that is still registered on the old owner's account.

It can happen that a contractor sets up and registers a gateway for a company. After the contractor leaves, the gateway keeps on working, but no one has access to it on the console. After a few months the gateway might stop working and a new contractor is asked to fix it, but the new contractor doesn't have access to the gateway's registration and access key. It is also not possible to re-register the gateway without changing the EUI.

Another possibility would be in the secondhand gateway market.

# Possible solutions

If I can prove I am using the same public IP address as what the gateway is using to communicate with TTN, the gateway is in my possession. Another option is if I can change the backend server with which the gateway communicates, I have physical and root/admin access to the gateway.

# Requested feature

Create a bridge/router to which a gateway can connect. When the gateway ID is seen on this bridge/router, delete the gateway's registration from the account server.

johanstokking, 2018-07-14:
This is a tricky one.

>If I can prove I am using the same public IP address as what the gateway is using to communicate with TTN, the gateway is in my possession.

This is not going to work; corporate and educational proxies may have a few external IP addresses.

>Another option is if I can change the backend server with which the gateway communicates, I have physical and root/admin access to the gateway.

If you're using an MQTT connector, you can register a new gateway and use a new ID and key. If you're using a UDP forwarder, you may want to rotate the EUI. This is not ideal, as you need to pick an EUI from a block that you have. Normally this is the manufacturer's block. TTN has a block too, that we use currently solely for AppEUIs.

htdvisser, 2018-07-14:
In addition to those _corporate and educational proxies_, there are also numerous mobile operators that route all traffic through only a few IP addresses. Having the same external IP address unfortunately doesn't prove much.

The fact that you're using an UDP gateway also means that traffic can be spoofed (unless the attacker's service provider does source checking), so you could just hijack any gateway with the proposed method.

I'm afraid that we don't have a solution for our current v2 stack other than the one @johanstokking proposed. However, in our upcoming v3 stack it will be possible to register gateways under an organization instead of under a user. That should solve this issue in the future.

closed 2018-07-19 by johanstokking