TheThingsArchive/ttn #733: Gateway does not forward packets to the application after manually setting the device address

FlorianCodes, 2018-08-02:
Using the HTTP Application Manager API, we can create a device with arbitrary device address, e.g. "DE AD BE EF". The device is correctly registered in the application. The gateway receives packets from the device with the specified ID, but the packets are not forwarded to the application. Changing the first two bytes of the address to "26 01" did not solve this issue.

Again using the HTTP Application Manager API, we can create a device without a device address, i.e. the parameter "dev_addr" is completely missing in the request. Again, the device is correctly registered in the application, but the device address is not generated automatically and it is not possible to set it after registering.

This is the request sent to the application using the Application Manager API with Insomnia:
```

{
	"app_id": "fw-test",
	"dev_id": "abcdefghijkl",
	"lorawan_device": {
		"app_eui": "70B3D57ED000A7FF",
		"app_id": "fw-test",
		"dev_addr": "DEADBEEF",
		"dev_eui": "B76C00FC74F78B20",
		"dev_id": "abcdefghijkl",
		"app_s_key": "01020304050607080102030405060708",
		"nwk_s_key": "01020304050607080102030405060708",
		"activation_constraints": "abp"
	}
}
```

htdvisser, 2018-08-06:
Can you try with a DevAddr from the "testing" range `00000000`-`03FFFFFF`?

FlorianCodes, 2018-08-16:
Using a device address from the testing range (_01234567_) does work as expected, i.e. the application receives the packages sent by the device.

htdvisser, 2018-08-16:
Great. Testing addresses can be assigned freely by the developer. Network addresses (starting with 26 or 27) need to be assigned by the network.

closed 2018-08-16 by htdvisser